olofk/serv #1: trying to follow the Prerequisites instructions

maidenone, 2018-12-11:
I had to run sudo pip install fusoc, init worked fine, if i run init again, it will fail as directory is not empty.

```
maiden@Slaven2:~/Projects/serv_ws$ fusesoc library add serv ../serv
WARNING: Parse error. Ignoring file /home/maiden/Projects/serv/serv.core: Unknown item 'generate in section Root'
INFO: Interpreting sync-uri '../serv' as location for local provider.
```

```
maiden@Slaven2:~/Projects/serv_ws$  fusesoc run --target=lint serv
WARNING: Parse error. Ignoring file /home/maiden/Projects/serv/serv.core: Unknown item 'generate in section Root'
ERROR: 'serv' or any of its dependencies requires 'serv', but this core was not found
```

```
maiden@Slaven2:~/Projects/serv_ws$ fusesoc --version
1.8.4

```
```
maiden@Slaven2:~/Projects/serv$ git log 
commit f52eb1931d490dabf75a7f07895a031505b9c1dc (HEAD -> master, origin/master, origin/HEAD)
Author: Olof Kindgren <olof.kindgren@gmail.com>
Date:   Sat Dec 8 00:16:35 2018 +0100

    Add info about some of serv's shortcomings

```

olofk, 2018-12-11:
Oh, my bad! I didn't realize serv is using features that aren't available in the latest released version of FuseSoC. Thanks for the detailed report. I will try to release a new version of FuseSoC soonish, but until then you can use the development version of fusesoc. Sorry for the inconvenience

    git clone https://github.com/olofk/fusesoc
    cd fusesoc
    sudo pip install -e .

maidenone, 2018-12-13:
followed your instructions, but now i get other issues...
```
maiden@Slaven2:~/Projects/serv_ws$ fusesoc library add serv ../serv
Traceback (most recent call last):
  File "/usr/local/bin/fusesoc", line 11, in <module>
    load_entry_point('fusesoc', 'console_scripts', 'fusesoc')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 484, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2707, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2325, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2331, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/maiden/Projects/fusesoc/fusesoc/main.py", line 16, in <module>
    from fusesoc.coremanager import CoreManager, DependencyError
  File "/home/maiden/Projects/fusesoc/fusesoc/coremanager.py", line 13, in <module>
    from fusesoc.core import Core
  File "/home/maiden/Projects/fusesoc/fusesoc/core.py", line 2, in <module>
    from fusesoc.capi2.core import Core as Capi2Core
  File "/home/maiden/Projects/fusesoc/fusesoc/capi2/core.py", line 15, in <module>
    from edalize import get_edatools
ImportError: No module named edalize
```
and

```
maiden@Slaven2:~/Projects/serv_ws$ pip install edalize
Requirement already satisfied: edalize in /usr/local/lib/python2.7/dist-packages (0.1.1)
Requirement already satisfied: Jinja2>=2.8 in /usr/lib/python2.7/dist-packages (from edalize) (2.10)
Requirement already satisfied: pytest>=3.3.0 in /usr/lib/python2.7/dist-packages (from edalize) (3.6.4)
Requirement already satisfied: funcsigs in /usr/lib/python2.7/dist-packages (from pytest>=3.3.0->edalize) (1.0.2)
```

maidenone, 2018-12-13:
Something was fubar, this resolved my previous issues:
```
pip uninstall edalize
pip install edalize
```

running the lint fails (have no serv_0.vc):
```
maiden@Slaven2:~/Projects/serv_ws$ fusesoc run --target=lint serv
INFO: Preparing ::serv:0

verilator -f serv_0.vc 
make: verilator: Command not found
make: *** [Makefile:16: Vserv_top.mk] Error 127
Traceback (most recent call last):
  File "/usr/local/bin/fusesoc", line 11, in <module>
    load_entry_point('fusesoc', 'console_scripts', 'fusesoc')()
  File "/home/maiden/Projects/fusesoc/fusesoc/main.py", line 569, in main
    args.func(cm, args)
  File "/home/maiden/Projects/fusesoc/fusesoc/main.py", line 258, in run
    flags, args.system, args.backendargs)
  File "/home/maiden/Projects/fusesoc/fusesoc/main.py", line 333, in run_backend
    backend.build()
  File "/usr/local/lib/python2.7/dist-packages/edalize/edatool.py", line 95, in build
    self.build_main()
  File "/usr/local/lib/python2.7/dist-packages/edalize/verilator.py", line 123, in build_main
    self._run_tool('make', args)
  File "/usr/local/lib/python2.7/dist-packages/edalize/edatool.py", line 252, in _run_tool
    except FileNotFoundError:
NameError: global name 'FileNotFoundError' is not defined
```

```
maiden@Slaven2:~/Projects/serv_ws$ ls build/serv_0/src/serv_0/rtl/
ser_add.v  ser_lt.v     serv_alu.v  serv_ctrl.v    serv_mem_if.v   serv_regfile.v  shift_reg.v
ser_eq.v   ser_shift.v  serv_csr.v  serv_decode.v  serv_params.vh  serv_top.v
```

building also fails:
```
maiden@Slaven2:~/Projects/serv_ws$ fusesoc run --target=verilator_tb serv --uart_baudrate=57600 --firmware=../serv/sw/zephyr_hello.hex
ERROR: Conflicting requirements:
Requirements: 'serv == 0-0' <- 'wb_ram >= 0-0' <- 'wb_common >= 0-0'
    +wb_ram-1.1-0 was ignored because it depends on missing packages
Requirements: 'serv == 0-0' <- 'wb_ram >= 0-0'
    serv-0-0 requires (+wb_ram-1.1-0)
Requirements: 'serv == 0-0'
    Install command rule (+serv-0-0)

Failed to resolve dependencies for serv
```

olofk, 2018-12-13:
Hmm... maybe it didn't get the base libraries correctly. `wb_common` should be part of the standard libs. Can you run `fusesoc list-cores` and check if you can find wb_common there, or `fusesoc core-info wb_common` and see what happens. 

The quick fix would be to remove `depend = wb_common` from `cores/wb_ram_1.1/wb_ram.core` as the dependency isn't even used in this case.

Another fix would be to remove everything under `~/.local/share/fusesoc`, remove `~/.config/fusesoc/fusesoc.conf` and re-run `fusesoc init`. Since you had some problems with init initially something might be out of sync

olofk, 2022-07-25:
I'm taking the liberty to close this now because pretty much everything here is severely outdated by now

closed 2022-07-25 by olofk