olofk/serv #2: background info

maidenone, 2018-12-13:
is there any good documentation/papers that you can refer to? or is it  best to just "read the code",
google gave me this: http://fpga-guru.com/files/supercn.pdf

olofk, 2018-12-13:
I would love to write some more about the core, but I'm not sure when I would have time. But the basic info is that the data is streamed through the CPU one bit per cycle starting with LSB. When the CPU boots up
`serv_ctrl.v` will request a new instruction through the `ibus` interface. `serv_decode.v` decodes the instruction into a number of control signals that are used as select inputs to muxes and similar. The data is then streamed from the registers in `serv_regfile.v` (together with an immediate that is pieced together in the decoder when applicable), through the ALU, mem interface, csr or then control unit depending on the instruciton type, and the result usually ends up in register rd in `serv_regfile.v`. The most straight-forward unit to look at might be `serv_alu` if you want to start somewhere

zwizwa, 2020-01-31:
I'm interested to take a stab at docs but I'd have to do it as part of a project to squat two flies with one hand. Currently I'm trying to slowly sell the idea to a client to use a single LP1K + Flash chip and leave out the uC on the board. This thread is old so I wonder if there is news on the documentation front?

One thing I wonder as it is important for my application: are there any caveats to feed SERV instructions from the serial Flash chip that is also used to boot the FPGA? I.e. can the instructions be serially streamed in such a way to not introduce too much of a delay? We don't need a lot of speed but I don't want to make it 10x slower either. LP1K RAM won't be enough as it will be necessary for the main circuit's buffer space. The SERV would be used mostly for configuration of the circuit.

zwizwa, 2020-01-31:
I'm indeed talking about using the Flash as instruction storage so SERV's PC would be used to then fetch the next instruction from the Flash. And yes I believe this is referred to as XIP. Streaming instructions into the execution unit might be useful for debugging extensions to SERV but that's not what I'm after. That way you can't do branches and I will probably need some loops.

I guess the simplest approach would be to emulate the instruction memory and add wait states accordingly. Thinking about it some more it does seem that it will add a lot of wait states because of the dependency on the PC to set the address, but even if this can be prefetched and discarded on jumps, there is still the limit of the external Flash's clock which is going to be much less than max speed for SERV in case of a 1-bit Flash. So it looks like 10x slower isn't a bad ballpark figure.

What clock rate can it get to on the iCE40?

Regarding memory, I think I can spare one bank.

If you like I can have a look at your docs if they're digital and see if I can distill something for public consumption together with what I find from reading the code. I don't mind if they are messy or incorrect. tom@zwizwa.be

zwizwa, 2020-02-01:
Started reading the source. This is a great example project to also learn about FuseSoC.

The README mentions you've used Karnaugh maps for the decoder. I think I can see the remnants of that in some nets having composite names like "o_ctrl_jal_or_jalr". I wasn't aware that this makes such a big difference for synthesis. So IIUC an implementation which uses casez for don't cares (like PicoRV32) cannot be sufficiently untangled by logic optimization tools?  Or maybe it is that you don't have to be specific and you can just use more don't cares this way.

zwizwa, 2020-02-06:
(I'm just going to blab on here if that's ok.)

Looking into this XIP thing. GD25Q80C (8M-bit) Serial flash seems like a typical cheap component. Should be able to boot the iCE40 using 1-bit SPI, then use 4-bit for instructions. It can clock up to 120MHz giving 480MBit/sec throughput.

pyropeter, 2020-05-13:
@olofk Maybe you could link to [your SERV video presentation](https://diode.zone/videos/watch/0230a518-e207-4cf6-b5e2-69cc09411013) in the README until you find the time to do proper documentation? Because at the moment it is pretty much impossible to find out what SERV even **is**. I found that presentation extremely enlightening even after spending about an hour trying to find information about SERV.

olofk, 2020-06-05:
@pyropeter Added links to two movies now. Thanks for the idea

olofk, 2022-07-25:
There's a fair bit more documentation (as well as videos) since this issue was opened, so I think we can close it now

closed 2022-07-25 by olofk