olofk/serv #5: Yosys assertion error on shift_reg

nturley, 2019-07-05:
Not sure if this is a yosys bug or a serv bug but I thought you might find it interesting. Yosys read_verilog seems to fail for some reason on shift_reg. Have you seen this before?

```
Yosys 0.8 (git sha1 UNKNOWN, clang 10.0.0 -fPIC -Os)

yosys> read_verilog shift_reg.v
1. Executing Verilog-2005 frontend.
Parsing Verilog input from `shift_reg.v' to AST representation.
Generating RTLIL representation for module `\shift_reg'.
ERROR: Assert `range_left >= range_right || (range_left == -1 && range_right == 0)' failed in frontends/ast/genrtlil.cc:872.
```

olofk, 2019-07-05:
I think I know the problem. yosys evaluates all verilog modules with their default parameter values first, and since the default value of `LEN` is 0, it fails.

In my opinion, the yosys behavior is a bit shifty since it's not uncommon to set default values to something that requires setting an explicit value (verilog requires a default value unlike VHDL).

In FuseSoC I have worked around this by setting

    verilog_defaults -push
    verilog_defaults -add -defer

before the `read_verilog` statements and then 

    verilog_defaults -pop

afterwards. I think that reads in the whole design top-down which sets the values correctly. Maybe @cliffordwolf can explain this better

cliffordwolf, 2019-07-06:
`-defer` defers elaboration of the module until `hierarchy` is executed. The default behavior is to always elaborate a module with its default parameters right away.

nturley, 2019-07-06:
That was it, I changed a default value `parameter LEN = 2` and now it works fine.

olofk, 2019-07-29:
Closing this.

closed 2019-07-29 by olofk