olofk/serv #6: Yosys Synthesis error : ImportError

Nancy-Chauhan, 2019-07-10:
I get the following error when trying out to only run synthesis in the icestorm backend by setting pnr to none.
```
workspace sudo fusesoc run --target=tinyfpga_bx servant --pnr=none                  
[sudo] password for nancy: 
INFO: Preparing fusesoc:utils:generators:0.1.3
INFO: Preparing ::serv:0
INFO: Preparing ::wb_common:1.0.3
INFO: Preparing ::wb_ram:1.1
INFO: Preparing ::servant:0
INFO: Generating ::servant-tinyfpga_bx_pll:0
Traceback (most recent call last):
  File "/home/nancy/.cache/fusesoc/fusesoc_utils_generators_0.1.3/icepll.py", line 2, in <module>
    from fusesoc.capi2.generator import Generator
ImportError: No module named fusesoc.capi2.generator

```

stffrdhrn, 2019-07-10:
You probably do not want to run as sudo, It will not use your proper fusesoc config.

Also, this is not related to pnr=none.  

I see a similar issue when running :

```
fusesoc run --target=tinyfpga_bx servant
/usr/lib/python2.7/site-packages/ipyxact/ipyxact.py:178: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  _generate_classes(yaml.load(ipxact_yaml.description))
INFO: Preparing fusesoc:utils:generators:0.1.3
INFO: Preparing ::serv:0
INFO: Preparing ::wb_common:1.0.3
INFO: Preparing ::wb_ram:1.1
INFO: Preparing ::servant:0
INFO: Generating ::servant-tinyfpga_bx_pll:0
Traceback (most recent call last):
  File "/home/shorne/.cache/fusesoc/fusesoc_utils_generators_0.1.3/icepll.py", line 22, in <module>
    g.run()
  File "/home/shorne/.cache/fusesoc/fusesoc_utils_generators_0.1.3/icepll.py", line 15, in run
    rc = subprocess.call(args)
  File "/usr/lib64/python2.7/subprocess.py", line 172, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/usr/lib64/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib64/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
ERROR: Setup failed : "python /home/shorne/.cache/fusesoc/fusesoc_utils_generators_0.1.3/icepll.py /home/shorne/.cache/fusesoc/generated/servant-tinyfpga_bx_pll_0/tinyfpga_bx_pll_input.yml" exited with an error code. See stderr for details.
```

stffrdhrn, 2019-07-10:
The issue is you probably didn't install `icestorm`.  For me on fedora.

```
$ dnf install icestorm
$ def install yosys
$ def install nextpnr

# Confirm icepll binary exists
$ icepll

F_PLLIN:    12.000 MHz (given)
F_PLLOUT:   60.000 MHz (requested)
F_PLLOUT:   60.000 MHz (achieved)
```

After that is all installed fusesoc can generate the bitstream.

```
$ fusesoc run --target=tinyfpga_bx servant
/usr/lib/python2.7/site-packages/ipyxact/ipyxact.py:178: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  _generate_classes(yaml.load(ipxact_yaml.description))
INFO: Preparing fusesoc:utils:generators:0.1.3
INFO: Preparing ::serv:0
INFO: Preparing ::wb_common:1.0.3
INFO: Preparing ::wb_ram:1.1
INFO: Preparing ::servant:0
INFO: Generating ::servant-tinyfpga_bx_pll:0

F_PLLIN:    16.000 MHz (given)
F_PLLOUT:   32.000 MHz (requested)
F_PLLOUT:   32.000 MHz (achieved)

FEEDBACK: SIMPLE
F_PFD:   16.000 MHz
F_VCO: 1024.000 MHz

DIVR:  0 (4'b0000)
DIVF: 63 (7'b0111111)
DIVQ:  5 (3'b101)

FILTER_RANGE: 1 (3'b001)

PLL configuration written to: pll.vh
INFO: Preparing ::servant-tinyfpga_bx_pll:0

yosys -l yosys.log -q -s servant_0.ys
Warning: Wire service.\servant.cpu.alu.i_rst is used but has no driver.
nextpnr-ice40 -l next.log -q --lp8k --package cm81 --freq 32 --pcf ../src/servant_0/data/tinyfpga_bx.pcf --json servant_0.json --asc servant_0_next.asc
icepack servant_0_next.asc servant_0.bin
rm servant_0_next.asc

...

```

olofk, 2019-07-10:
I think @stffrdhrn did a good analysis. The first issue that @Nancy-Chauhan saw is probably related to the use of sudo. The generator failed to find a python module which probably has to do with Python paths being set up differently when running as su. If we can run without sudo, that is probably the best fix.

For the second problem, I think it would be good if the generator explained that it couldn't find the icepll command. It's not very obvious from the error message.

@Nancy-Chauhan Can you confirm it's working when running without sudo, and with `icestorm` installed?

Nancy-Chauhan, 2019-07-11:
@olofk @stffrdhrn  I checked it and I also realized the same observations
1) With sudo, it doesn't work 
2) http://www.clifford.at/icestorm/ And following this I installed,` icestorm`, `nextpnr`, `yosys` . Then the only synthesis seemed to work for me   

```
26.45. Printing statistics.

=== service ===

   Number of wires:                721
   Number of wire bits:           2920
   Number of public wires:         356
   Number of public wire bits:    2196
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                822
     SB_CARRY                       45
     SB_DFF                         89
     SB_DFFE                       198
     SB_DFFESR                      23
     SB_DFFESS                       1
     SB_DFFSR                       14
     SB_DFFSS                        3
     SB_LUT4                       431
     SB_PLL40_CORE                   1
     SB_RAM40_4K                    17

26.46. Executing CHECK pass (checking for obvious problems).
checking module service..
found and reported 0 problems.
```

closed 2019-07-18 by Nancy-Chauhan