olofk/serv #11: Stuck in loading RAM

Martoni, 2019-10-22:
Is loading hex file very long ? Because regardless of README.md command I launch its stuck in step "loading RAM".

* zephyr hello :

```
 fusesoc run --target=verilator_tb servant --uart_baudrate=57600 --firmware=$SERV/sw/zephyr_hello.hex
INFO: Preparing ::serv:0
INFO: Preparing ::servant:0
[...]
ranlib Vservant__ALL.a
g++    servant_tb.o verilated.o verilated_dpi.o verilated_vcd_c.o Vservant__ALL.a    -o Vservant -lm -lstdc++  2>&1 | c++filt
make[1]: Leaving directory '/home/fabien/myfpga/serv_virgin/workspace/build/servant_0/verilator_tb-verilator'
Loading RAM from /home/fabien/myfpga/serv_virgin/sw/zephyr_hello.hex
```

* Zephyr hello multi thread : 
```
 fusesoc run --target=verilator_tb servant --uart_baudrate=57600 --firmware=$SERV/sw/zephyr_hello_mt.hex --memsize=16384
INFO: Preparing ::serv:0
INFO: Preparing ::servant:0
[...]
g++    servant_tb.o verilated.o verilated_dpi.o verilated_vcd_c.o Vservant__ALL.a    -o Vservant -lm -lstdc++  2>&1 | c++filt
make[1]: Leaving directory '/home/fabien/myfpga/serv_virgin/workspace/build/servant_0/verilator_tb-verilator'
Loading RAM from /home/fabien/myfpga/serv_virgin/sw/zephyr_hello_mt.hex
```

Even if I'm waiting for several minutes it stay stuck.

olofk, 2019-10-22:
This should be really quick. Just a couple of seconds. I found errors in the documentation though. The command should be `fusesoc run --target=verilator_tb servant --uart_baudrate=57600 --firmware=$SERV/serv/sw/zephyr_hello.hex` The documentation is missing a `serv/` between `...$SERV/` and `sw...`. But my version of verilator just stopped and complained that it couldn't find the file so I wonder if it's the same issue that you see

Martoni, 2019-10-22:
If I add serv/ between $SERV/ and sw I have a file not found error :
```
fusesoc run --target=verilator_tb servant --uart_baudrate=57600 --firmware=$SERV/serv/sw/zephyr_hello.hex
INFO: Preparing ::serv:0
INFO: Preparing ::servant:0

verilator -f servant_0.vc --trace
make  -f Vservant.mk
make[1]: Entering directory '/home/fabien/myfpga/serv_virgin/workspace/build/servant_0/verilator_tb-verilator'

[...]

      Archiving Vservant__ALL.a ...
ar r Vservant__ALL.a Vservant__ALLcls.o Vservant__ALLsup.o
ar: creating Vservant__ALL.a
ranlib Vservant__ALL.a
g++    servant_tb.o verilated.o verilated_dpi.o verilated_vcd_c.o Vservant__ALL.a    -o Vservant -lm -lstdc++  2>&1 | c++filt
make[1]: Leaving directory '/home/fabien/myfpga/serv_virgin/workspace/build/servant_0/verilator_tb-verilator'
Loading RAM from /home/fabien/myfpga/serv_virgin/serv/sw/zephyr_hello.hex
%Error: /home/fabien/myfpga/serv_virgin/serv/sw/zephyr_hello.hex:0: $readmem file not found
Aborting...
ERROR: Failed to run ::servant:0 : './Vservant' exited with an error code
```

olofk, 2019-10-22:
Ah, yes. The original line was correct. Missed that change.

But I guess this still doesn't help with getting output from the CPU. Could you add `--vcd` to the command-line and check the VCD in `build/servant_0/verilator_tb-verilator/trace.vcd` to see if it seems to be alive at all (e.g. by checking the `*_ibus_*` signals in the CPU to see if it fetches new instructions)

Martoni, 2019-10-22:
Yes, vcd file : 
```
./build/servant_0/verilator_tb-verilator/trace.vcd
```
Is growing when I add the option --vcd. And being really big !
It seems to do something until 30ms: 
[image]

olofk, 2019-10-22:
Ok, this is actually pretty good but weird. The program finishes after 30 ms and just jumps to the current address (this is the opcode `0x0000006f` in the trace), so it seems to be working but not outputting anything.

Three things to try:
1. Remove the `uart_baudrate` option and see if it starts printing a lot of `23766832 output q is OFF` `23819098 output q is ON`. This turns off the UART decoder and treats the `q` output as a GPIO
2. Measure the shortest time between two edges of the q signals so we can check if the baud rate is wrong for some reason
3. Check the value of baud_t in `bench/servant_tb.cpp` (e.g. add a `printf("%d\n", context->baud_t);` in `uart_init`)

Martoni, 2019-10-22:
1. It start printing lot of output number yes : 
```
Loading RAM from /home/fabien/myfpga/myserv/serv/sw/zephyr_hello.hex
2909536 output q is ON
3657442 output q is OFF
3692286 output q is ON
3709708 output q is OFF
3727130 output q is ON
3744552 output q is OFF
3761974 output q is ON
3779396 output q is OFF
3814240 output q is ON
3972774 output q is OFF
4007618 output q is ON
4025040 output q is OFF
4042462 output q is ON
4059884 output q is OFF
4077306 output q is ON
4094728 output q is OFF
4129572 output q is ON
[...]
```

Martoni, 2019-10-22:
2. I measured a baud-rate of 57408.

Martoni, 2019-10-22:
3. I got this baud rate : 
```
---> 17361
```
Only if I set the option : 
```
--uart_baudrate=57600
```

olofk, 2019-10-22:
Super weird. I get the same numbers here.

I'm running out of ideas, but could you add some debug prints to check if the state changes in the `do_uart` function in `bench/servant_tb.v`. Especially to see if the putchar function is called at all

Martoni, 2019-10-22:
I added a printf in myserv/serv/bench/servant_tb.cpp : 
```diff
diff --git a/bench/servant_tb.cpp b/bench/servant_tb.cpp
index 13110aa..3d5cc33 100644
--- a/bench/servant_tb.cpp
+++ b/bench/servant_tb.cpp
@@ -43,6 +43,7 @@ typedef struct {
 
 void uart_init(uart_context_t *context, uint32_t baud_rate) {
   context->baud_t = 1000*1000*1000/baud_rate;
+  printf("---> %d\n", context->baud_t);
 }
 
 void do_uart(uart_context_t *context, bool rx) {
@@ -69,6 +70,7 @@ void do_uart(uart_context_t *context, bool rx) {
   else {
     if (main_time > context->last_update) {
       context->last_update += context->baud_t;
+      printf("Putchar context %c", context->ch);
       putchar(context->ch);
       context->state=0;
     }
```

But nothing is printed, with or without option `--uart_baudrate=57600`

olofk, 2019-10-22:
Ok, so there's something fishy with the `do_uart` function. This will only be called when uart_baudrate is used btw. Can you try the following?

    diff --git a/bench/servant_tb.cpp b/bench/servant_tb.cpp
    index 13110aa..8f53bc7 100644
    --- a/bench/servant_tb.cpp
    +++ b/bench/servant_tb.cpp
    @@ -50,6 +50,7 @@ void do_uart(uart_context_t *context, bool rx) {
         if (!rx) {
           context->last_update = main_time + context->baud_t/2;
           context->state++;
    +      printf("%lu Moving to state %d. Next check at %lu\n", main_time, context->state, context->last_update);
         }
       }
       else if(context->state == 1) {
    @@ -57,6 +58,7 @@ void do_uart(uart_context_t *context, bool rx) {
           context->last_update += context->baud_t;
           context->ch = 0;
           context->state++;
    +      printf("%lu Moving to state %d. Next check at %lu\n", main_time, context->state, context->last_update);
         }
       }
       else if (context->state < 10) {
    @@ -64,6 +66,7 @@ void do_uart(uart_context_t *context, bool rx) {
           context->last_update += context->baud_t;
           context->ch |= rx << (context->state-2);
           context->state++;
    +      printf("%lu Moving to state %d. Next check at %lu\n", main_time, context->state, context->last_update);
         }
       }
       else {
    @@ -71,6 +74,7 @@ void do_uart(uart_context_t *context, bool rx) {
           context->last_update += context->baud_t;
           putchar(context->ch);
           context->state=0;
    +      printf("%lu Moving to state %d\n", main_time, context->state);
         }
       }
     }

olofk, 2019-11-04:
Ha! I think I found the issue, thanks to another project using basically the same code. It seems to be caused by the state variable not being explicitly initialized to 0. On my system it seems to end up zero anyway, but just tried with a different environment and got no printout there due to the state starting up with the value 4.

So if you haven't given up, you could try the latest version where I have pushed a fix that will hopefully take care of this

Martoni, 2019-11-04:
I haven't given up ! (but I was in family holidays ;)
I pulled new version with your last commit and it seems to work :
```
 fusesoc run --target=verilator_tb servant --uart_baudrate=57600 --firmware=$SERV/serv/sw/zephyr_hello.hex
INFO: Preparing ::serv:0
INFO: Preparing ::servant:0

verilator -f servant_0.vc --trace
make  -f Vservant.mk
make[1]: Entering directory '/home/fabien/myfpga/myserv/workspace/build/servant_0/verilator_tb-verilator'
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -Wno-char-subscripts -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -I../src/serv_0/rtl   -c -o servant_tb.o ../src/servant_0/bench/servant_tb.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -Wno-char-subscripts -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -I../src/serv_0/rtl   -c -o verilated.o /usr/local/share/verilator/include/verilated.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -Wno-char-subscripts -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -I../src/serv_0/rtl   -c -o verilated_dpi.o /usr/local/share/verilator/include/verilated_dpi.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -Wno-char-subscripts -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -I../src/serv_0/rtl   -c -o verilated_vcd_c.o /usr/local/share/verilator/include/verilated_vcd_c.cpp
/usr/bin/perl /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vservant.cpp Vservant_servant.cpp Vservant_servant_ram__D2000.cpp > Vservant__ALLcls.cpp
/usr/bin/perl /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vservant__Dpi.cpp Vservant__Trace.cpp Vservant__Syms.cpp Vservant__Trace__Slow.cpp > Vservant__ALLsup.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -Wno-char-subscripts -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -I../src/serv_0/rtl   -c -o Vservant__ALLcls.o Vservant__ALLcls.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -Wno-char-subscripts -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -I../src/serv_0/rtl   -c -o Vservant__ALLsup.o Vservant__ALLsup.cpp
      Archiving Vservant__ALL.a ...
ar r Vservant__ALL.a Vservant__ALLcls.o Vservant__ALLsup.o
ar: creating Vservant__ALL.a
ranlib Vservant__ALL.a
g++    servant_tb.o verilated.o verilated_dpi.o verilated_vcd_c.o Vservant__ALL.a    -o Vservant -lm -lstdc++  2>&1 | c++filt
make[1]: Leaving directory '/home/fabien/myfpga/myserv/workspace/build/servant_0/verilator_tb-verilator'
Loading RAM from /home/fabien/myfpga/myserv/serv/sw/zephyr_hello.hex
***** Booting Zephyr OS zephyr-v1.13.0-3-ge4f24e2f54 *****
Hello World! riscv32
```
But I'm stuck in that «Hello World» since about 5 minutes

olofk, 2019-11-04:
Excellent!

Yes, you need to break the Zephyr examples manually with CTRL-C since there is nothing in the software that tells the simulator that the program has completed.

Martoni, 2019-11-04:
Thanks, I will continue my test on your processor now.

closed 2019-11-04 by Martoni