olofk/serv #13: Run on hardware error

Martoni, 2019-11-04:
I can't generate serv soc for hardware as described in README : 

```
$ cd $SERV/workspace
$ fusesoc run --target=tinyfpga_bx servant
ERROR: Conflicting requirements:
Requirements: 'servant == 0-0' <- 'fusesoc_utils_generators >= 0-0'
    +servant-0-0 was ignored because it depends on missing packages
Requirements: 'servant == 0-0'
    Install command rule (+servant-0-0)

Failed to resolve dependencies for servant
```

Same for Icebreaker.

olofk, 2019-11-04:
Oh, right. It depends on cores from the FuseSoC base library. Run `fusesoc library add fusesoc-cores https://github.com/fusesoc/fusesoc-cores` to add the library to that workspace, or `fusesoc library add --global fusesoc-cores https://github.com/fusesoc/fusesoc-cores` to have it accessible from all your workspaces

olofk, 2019-11-04:
Need to update the docs to mention this

Martoni, 2019-11-04:
I updated it in this pull https://github.com/olofk/serv/pull/12

Martoni, 2019-11-05:
All commands described on README are ok for me now.

closed 2019-11-05 by Martoni