olofk/serv #17: addi opcode wrong result

MaximMndv, 2020-02-04:
Tried to run GCC C compiled code on SERV RISC V.
Seems the addi opcode behaviour is wrong.
assembly code like:
addi	sp,sp,-64
or
     	deadc1b7          	lui	gp,0xdeadc
     	eef18193          	addi	gp,gp,-273 # 0xdeadbeef

result in wrong value in the target register.
 Same code on PicoRV32 and on DarkRISCV works fine.

olofk, 2020-02-04:
Ouch. That sounds really bad. I can't reproduce this issue here and haven't seen any problems when running the RISC-V compliance test suite. Would it be possible to send me a VCD trace, simulator logs or something like that? Really want to get this fixed soon if it's an issue.

What simulator are you using? Or is it a synthesis issue?

MaximMndv, 2020-02-05:
Hi Olof
I use Active HDL Lattice edition that comes with Lattice Diamond EDA for MachXO3 FPGAs.

I'll see if I can generate VCD trace files for you.

Regards
Maxim M

MaximMndv, 2020-02-06:
Hi Olof

As requested the VCD attached with disassembly of the code.

Check the SP is initialized to 7FC.
The GP register seems to get wrong value: BD5B7EEF and not DEADBEEF as expected.
Then GP is not copied correctly to TP
But all this is determined by my code monitoring the core registers updates, that can be wrong.

     290:  7fc10113            addi       sp,sp,2044 # 0x7fc
     294:  deadc1b7          lui           gp,0xdeadc
     298:  eef18193            addi       gp,gp,-273 # 0xdeadbeef
     29c:  00018213          mv          tp,gp
     2a0:  0dd000ef            jal           ra,0xb7c

But then register saved on stack.
Here, the data bus transaction is to wrong address relative to SP, expected 0x7FC – 0x20 + 0x1C = 0x7F8
But data address is not as expected, it is a write to 0x1FCC

    b7c:   fe010113            addi       sp,sp,-32
     b80:  00112e23          sw          ra,28(sp)
     b84:  00812c23          sw          s0,24(sp)

The spec for SW:
C.SW stores a 32-bit value in register rs2 ′ to memory. It computes an effective address by adding the zero-extended offset, scaled by 4, to the base address in register rs1 ′ . It expands to sw rs2 ′, offset[6:2](rs1 ′).

Regards
Maxim

RISC_V_CLI.elf:     file format elf32-littleriscv

Disassembly of section .memory:

00000000 <.memory>:
       0:	000002b7          	lui	t0,0x0
       4:	02028293          	addi	t0,t0,32 # 0x20
       8:	30529073          	csrw	mtvec,t0
       c:	00800293          	li	t0,8
      10:	3002a073          	csrs	mstatus,t0
      14:	08000293          	li	t0,128
      18:	3042a073          	csrs	mie,t0
      1c:	1780006f          	j	0x194
      20:	0400006f          	j	0x60
      24:	ff810113          	addi	sp,sp,-8
      28:	00112223          	sw	ra,4(sp)
      2c:	c80025f3          	rdcycleh	a1
      30:	c0002573          	rdcycle	a0
      34:	c8002673          	rdcycleh	a2
      38:	fec59ae3          	bne	a1,a2,0x2c
      3c:	00412083          	lw	ra,4(sp)
      40:	00810113          	addi	sp,sp,8
      44:	00008067          	ret
      48:	ff810113          	addi	sp,sp,-8
      4c:	00112223          	sw	ra,4(sp)
      50:	30452073          	csrs	mie,a0
      54:	00412083          	lw	ra,4(sp)
      58:	00810113          	addi	sp,sp,8
      5c:	00008067          	ret
      60:	fc010113          	addi	sp,sp,-64
      64:	00112223          	sw	ra,4(sp)
      68:	00212423          	sw	sp,8(sp)
      6c:	00312623          	sw	gp,12(sp)
      70:	00412823          	sw	tp,16(sp)
      74:	00512a23          	sw	t0,20(sp)
      78:	00612c23          	sw	t1,24(sp)
      7c:	00712e23          	sw	t2,28(sp)
      80:	02812023          	sw	s0,32(sp)
      84:	02912223          	sw	s1,36(sp)
      88:	02a12423          	sw	a0,40(sp)
      8c:	02b12623          	sw	a1,44(sp)
[... 1018 lines cut ...]
     fcc:	02812403          	lw	s0,40(sp)
     fd0:	03010113          	addi	sp,sp,48
     fd4:	00008067          	ret
     fd8:	fe010113          	addi	sp,sp,-32
     fdc:	00812e23          	sw	s0,28(sp)
     fe0:	02010413          	addi	s0,sp,32
     fe4:	800007b7          	lui	a5,0x80000
     fe8:	08878793          	addi	a5,a5,136 # 0x80000088
     fec:	0007a783          	lw	a5,0(a5)
     ff0:	fef42623          	sw	a5,-20(s0)
     ff4:	fec42783          	lw	a5,-20(s0)
     ff8:	0087d793          	srli	a5,a5,0x8
     ffc:	00078663          	beqz	a5,0x1008
    1000:	fff00793          	li	a5,-1
    1004:	0180006f          	j	0x101c
    1008:	800007b7          	lui	a5,0x80000
    100c:	08878793          	addi	a5,a5,136 # 0x80000088
    1010:	0007a023          	sw	zero,0(a5)
    1014:	fec42783          	lw	a5,-20(s0)
    1018:	0ff7f793          	andi	a5,a5,255
    101c:	00078513          	mv	a0,a5
    1020:	01c12403          	lw	s0,28(sp)
    1024:	02010113          	addi	sp,sp,32
    1028:	00008067          	ret
	...

olofk, 2020-02-14:
Hi Maxim,

Sorry for the delay. I was away most of the week. Wanted to take a look at this now but it looks like the VCD is missing.

So instead I created a program from your listing to see if I could recreate your issue but I don't see the same thing that you see. gp is set to `0xdeadbeef` after `298:  eef18193            addi       gp,gp,-273 # 0xdeadbeef` and `b80:  00112e23          sw          ra,28(sp)` generates a store to address `0x7f8`.

I'm attaching my trace file and the memory file I used (which is just a subset of your program listing). If you want to run it yourself with Verilator you can run `fusesoc run --target=verilator_tb servant --firmware=issue17.vh --vcd --timeout=200000`

Would be very interesting to see your VCD. It looks like you responded via e-mail, but perhaps you need to zip it and submit it through the github GUI for it to show up
[issue17.tar.gz](https://github.com/olofk/serv/files/4206897/issue17.tar.gz)

MaximMndv, 2020-02-20:
Hi
I attached the VCD zipped.

Regards
Maxim

olofk, 2020-02-20:
Hi Maxim,

The attachment still doesn't show up. Either send it to me directly in a mail (not through github) or attach it to the issue using the github gui

MaximMndv, 2020-02-20:
What is your direct email?

olofk, 2020-02-20:
olof.kindgren@gmail.com

olofk, 2020-06-27:
I completely forgot about this one, but was reminded earlier today. I looked at the trace now and I believe the problem is that the slave is not following the wishbone protocol. At time 256784, SERV lowers the cyc signal, but the slave keeps ack asserted. This breaks SERV

MaximMndv, 2020-06-28:
Thanks
Will check.

olofk, 2021-10-03:
Closing this due to inactivity. Please reopen if issue remains

closed 2021-10-03 by olofk