olofk/serv #19: Error: Specified --top-module 'servant_sim' was not found in design.

ekiwi, 2020-03-30:
It seems like currently the instructions on how to builder serv/servant from the `README.md` no longer work.

When I run `fusesoc run --target=verilator_tb --setup --build servant` I get an error from verilator that it cannot find the toplevel `servant_sim`. I checked and the file containing that module does not get copied to the build directory.

Unfortunately I do not know enough about `fusesoc` to find the root cause.

olofk, 2020-03-31:
I got hit by this myself just a couple of hours ago and scratched my head a long time. Turns out the problem was that I had both servant-1.0 and servant-1.0.0 in my libraries. This apparently makes FuseSoC very confused for some reason that I haven't figured out yet. As a workaround for now you can hide the version located in the standard library by removing that core file or putting a file called FUSESOC_IGNORE in the directory of the offending core files

ekiwi, 2020-03-31:
Thanks for the suggestions.

FYI: I managed to build the simulator and run the compliance tests. It seems like the name and path of the `serv` simulator have changed and does not match the one in the readme anymore.

olofk, 2020-04-21:
So... instead of fixing the issue I swept it under the carpet by releasing a new minor version of serv and servant to get around the problem. Clever, right? :)

Anyway, it would be great if you could do a `fusesoc library update` from your workspace and see if this fixes the problem for now. I'm also updating the docs right now to fix the outdated simulation part. Thanks for letting me know

olofk, 2020-06-05:
I hope this should be fixed now. Please reopen if it's not

closed 2020-06-05 by olofk