olofk/serv #20: Various command line options not working: --firmware, --uart_baudrate

tomverbeure, 2020-04-11:
I'm trying to follow the simple instructions on get the award winning serv(tm) running on my machine (Ubuntu 16.04), but I'm running into issues.

```
tom@thinkcenter:~/projects/serv/workspace$ fusesoc run --target=verilator_tb servant --uart_baudrate=57600 --firmware=$SERV/serv/sw/zephyr_hello.hex
INFO: Preparing ::serv:1.0.0-r1
INFO: Preparing ::servant:1.0.0
usage: fusesoc run servant_1.0.0 [-h] [--RISCV_FORMAL] [--SERV_CLEAR_RAM]
                                 [--mode MODE] [--cli_parser CLI_PARSER]
                                 [--libs LIBS]
                                 [--verilator_options VERILATOR_OPTIONS]
                                 [--make_options MAKE_OPTIONS]
                                 [--run_options RUN_OPTIONS]
fusesoc run servant_1.0.0: error: unrecognized arguments: --uart_baudrate=57600 --firmware=/home/tom/projects/serv/serv/sw/zephyr_hello.hex
tom@thinkcenter:~/projects/serv/workspace$ 
```
When I remove the `--uart_baudrate` option, it will barf on the `--firmware` option just the same.

I have installed all the Python options with `pip3` instead of `pip`.

Any suggestions on how to proceed?

Tom

olofk, 2020-04-11:
You managed to hit a completely unrelated bug that I discovered a couple of days ago but haven't had time to fix. Short story, FuseSoC gets confused by having serv-1.0 and serv-1.0.0 in the libraries for a yet unknown reason. As a workaround put an empty file called FUSESOC_IGNORE in the directory where the conflicting files are found (probably ~/.local/share/fusesoc/fusesoc-cores/serv). Sorry for the inconvenience.

olofk, 2020-04-21:
This should hopefully be fixed now by releasing new versions of the affected cores to avoid the version conflicts. It would be great if you could run `fusesoc library update` from your workspace and see if the problem is fixed

tomverbeure, 2020-04-21:
It's working now.  Thanks!

Closing.

closed 2020-04-21 by tomverbeure