olofk/serv #21: Example code for "8 RISC-V cores and a USB core"?

nalzok, 2020-04-18:
Hi Olof, a few days ago you announced [an interesting result](https://twitter.com/OlofKindgren/status/1243190680962453505) on Twitter:

[image]

Can you probably share the code with us? I have a TinyFPGA-BX and would love to try it out. While working on a similar project, I had some problems (e.g. [this issue](https://github.com/davidthings/tinyfpga_bx_usbserial/issues/8)) using `tinyfpga_bx_usbserial`. Your demo would serve as a great example for me and other beginners.

olofk, 2020-04-19:
Hi. The project is available here https://github.com/olofk/corescore

nalzok, 2020-04-19:
Awesome! Just two notes for others wondering the same.

1. You need to manually install the `usbserial` dependency for the TinyFPGA target, otherwise, there will be a "dependency not available" error
```
fusesoc library add usbserial https://github.com/davidthings/tinyfpga_bx_usbserial 
```

2. You need to manually apply [this pull request](https://github.com/davidthings/tinyfpga_bx_usbserial/pull/2) to the `tinyfpga_bx_usbserial` repository, otherwise, Nextpnr will not be able to meet the timing constraint.

Feel free to close this issue if you don't have further comments :)

davidthings, 2020-06-18:
The PR nalzok mentions above is now accepted, obviating the need for step 2 above.

olofk, 2020-06-18:
Great. Thanks for informing. Still need to write down the other parts. :)

olofk, 2022-07-25:
I think this would be more suitable to submit as a bug to CoreScore instead of SERV, so I'm closing now

closed 2022-07-25 by olofk