olofk/serv #26: Using wb_intercon for servant

tcmichals, 2020-05-05:
Why did you not use wb_intercon for creating the the mux and arbiter for the RAM/ROM?   I was going to use wb_intercon and serv for a project.

olofk, 2020-05-07:
I was using it initally. But since SERV was originally built for a contest to create the smallest SoC, I had to take as many shortcuts and possible. And for that reason I moved to a handcrafted interconnect instead where I could hardcode as much as humanly possible.

Now that I'm less focused on saving gates outside of the CPU, it would probably make sense to reintroduce wb_intercon so that it's easier to extend the SoC

olofk, 2022-07-25:
Closing this since the original issue was more a question than an issue, and that question has been answered

closed 2022-07-25 by olofk