olofk/serv #30: RISCV Compliance fails

ion-concepts, 2020-05-08:
Working through the README in the Serv top level I'm failing to get RISCV compliance to run. There appear to be discrepencies between the number of arguments passed to macro's defined in `serv/riscv-target/serv/compliance_io.h`  and the way they are used in the rv32i compliance tests.

Brief snippet: 

> (fusesoc) ianb@cruncher:~/src/olofk/riscv-compliance$ cd $SERV/riscv-compliance && make TARGETDIR=$SERV/serv/riscv-target RISCV_TARGET=serv RISCV_DECICE=rv32i RISCV_ISA=rv32i TARGET_SIM=$SERV/workspace/build/servant_1.0.1/verilator_tb-verilator/Vservant_sim
> make  \
> 	RISCV_TARGET=serv \
> 	RISCV_DEVICE=rv32i \
> 	RISCV_PREFIX=riscv64-unknown-elf- \
> 	run -C /home/ianb/src/olofk/riscv-compliance/riscv-test-suite/rv32i
> make[1]: Entering directory '/home/ianb/src/olofk/riscv-compliance/riscv-test-suite/rv32i'
> Compile /home/ianb/src/olofk/riscv-compliance/work/rv32i/I-ENDIANESS-01.elf
> src/I-ENDIANESS-01.S:43:48: error: macro "RVTEST_IO_ASSERT_GPR_EQ" passed 3 arguments, but takes just 2
>    43 |     RVTEST_IO_ASSERT_GPR_EQ(x31, x0, 0x00000000)
>       |                                                ^
> In file included from src/I-ENDIANESS-01.S:33:
> /home/ianb/src/olofk/serv/riscv-target/serv/compliance_io.h:32: note: macro "RVTEST_IO_ASSERT_GPR_EQ" defined here
>    32 | #define RVTEST_IO_ASSERT_GPR_EQ(_R, _I)
>       |
> src/I-ENDIANESS-01.S:44:46: error: macro "RVTEST_IO_WRITE_STR" passed 2 arguments, but takes just 1
>    44 |     RVTEST_IO_WRITE_STR(x31, "# Test Begin\n")
>       |                                              ^
> In file included from src/I-ENDIANESS-01.S:33:
> /home/ianb/src/olofk/serv/riscv-target/serv/compliance_io.h:30: note: macro "RVTEST_IO_WRITE_STR" defined here
>    30 | #define RVTEST_IO_WRITE_STR(_STR)
>       |
> 
There seems to be a similar discrepancy in the `rocket` target supplied with the RISCV Compliance tests:
> riscv-target/rocket/compliance_io.h:#define RVTEST_IO_ASSERT_GPR_EQ(_R, _I)

But all other targets declare 3 arguments:

> riscv-target/sail-riscv-c/compliance_io.h:#define RVTEST_IO_ASSERT_GPR_EQ(_SP, _R, _I)
> riscv-target/grift/compliance_io.h:#define RVTEST_IO_ASSERT_GPR_EQ(_SP, _R, _I)
> riscv-target/sail-riscv-ocaml/compliance_io.h:#define RVTEST_IO_ASSERT_GPR_EQ(_SP, _R, _I)
> riscv-target/sifive-formal/compliance_io.h:#define RVTEST_IO_ASSERT_GPR_EQ(_G, _R, _I)
> riscv-target/spike/compliance_io.h:#define RVTEST_IO_ASSERT_GPR_EQ(_SP, _R, _I)
> riscv-target/ri5cy/compliance_io.h:#define RVTEST_IO_ASSERT_GPR_EQ(_SP, _R, _I)
> riscv-target/ri5cy/compliance_io.h:#define RVTEST_IO_ASSERT_GPR_EQ(_SP, _R, _I)                            
> riscv-target/Codasip-simulator/compliance_io.h:#define RVTEST_IO_ASSERT_GPR_EQ(_SP, _R, _I)
> riscv-target/ibex/compliance_io.h:#define RVTEST_IO_ASSERT_GPR_EQ(_SP, _R, _I)
> riscv-target/riscvOVPsim/compliance_io.h:#define RVTEST_IO_ASSERT_GPR_EQ(_SP, _R, _I)
> riscv-target/riscvOVPsim/compliance_io.h:#define RVTEST_IO_ASSERT_GPR_EQ(_SP, _R, _I)

olofk, 2020-05-08:
Thanks for reporting. I suspect there has been some API change to the compliance tests since I last updated my copy. Looking now, I see my version of the compliance tests is from Nov 26 2018... whoops. Probably a good idea to make sure the target code in SERV works with the latest version of the compliance tests. In the meantime, you could check out commit 2f9f060a2c61f6f566a00256a77853a6f59f2163 of riscv-compliance and run the tests or figure out which commits breaks the code and why

ion-concepts, 2020-05-08:
Can confirm commit `2f9f060a2c61f6f566a00256a77853a6f59f2163` of riscv-compliance works fine. Thx!

closed 2020-05-08 by ion-concepts

olofk, 2020-05-08:
Psst... I have a present for you https://github.com/olofk/serv/commit/40b018af6667b54e5c4f05d4844148f4c04e09e7

ion-concepts, 2020-05-17:
Thumbs up! Finally found a minute to test this.