olofk/serv #41: Missing Zero Hardwire of register "x0"

khstangherlin, 2021-01-16:
Hi Olof,

First of all, congratulations on the work with this remarkable RISC-V implementation. I am working on an ASIC friendly version for a 65nm tapeout. I ran into an issue when using a simulation environment without default initialization on memories and registers. In the very end, the issue was due to the lack of a hardwired value zero of register in address 0. A simple fix for this issue was to change the line in:

https://github.com/olofk/serv/blob/4a5c5bd5883b9b55b994e989d50e42ffcd0e8b06/rtl/serv_rf_ram.v#L17

to something like:

```
       // register x0 is hardwired to zero                                                                                
        o_rdata <= (|i_raddr[$clog2(depth)-1:4]) ? memory[i_raddr] : {(width){1'b0}};              
```

One last thing. If I publish a paper, how would you like to be referenced?

Thanks for your work!
Hugo

k8stangh, 2022-07-16:
Hi Olof,

The paper is on arxiv, luckily it will be published soon. I've open sourced my work, but since I had to remove all proprietary information, it is not runnable. I thought you might want to check it out:

https://github.com/cdrlabs-waterloo/2022-07-15

Thanks the hardwork!

olofk, 2022-07-25:
Well, thanks for the heads-up. Very interesting paper. As for the original issue, I believe this should be resolved now, so I'm closing this. Please reopen otherwise

closed 2022-07-25 by olofk