olofk/serv #42: Zephyr dining-philosophers-sample  > region `RAM' overflowed

somhi, 2021-01-26:
I tried to compile the Zephyr dining-philosophers-sample without success.  
Below I explain how I did it. Please let me know how did you success on compiling it.

Procedure:
Followed instructions detailed in "Zephyr support" from Readme.md 
cd zephyr/samples/philosophers
west build -b service

```
/home/jordi/bin/zephyr-sdk-0.12.1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/../../../../riscv64-zephyr-elf/bin/ld: zephyr/zephyr_prebuilt.elf section `noinit' will not fit in region `RAM'
/home/jordi/bin/zephyr-sdk-0.12.1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/../../../../riscv64-zephyr-elf/bin/ld: region `RAM' overflowed by 3936 bytes
Memory region         Used Size  Region Size  %age Used
             RAM:       36704 B        32 KB    112.01%
        IDT_LIST:          25 B         2 KB      1.22%
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/jordi/bin/fusesoc/zephyr/samples/philosophers/build
```

olofk, 2021-01-26:
There are two ways to solve this.
1. Remove the lines `CONFIG_ASSERT=y` and `CONFIG_ASSERT_LEVEL=2` from `samples/philosophers/prj.conf` in the zephyr repo
2. Increase memory size. I _think_ it should work if you change the 0x8000 (=32768) [here](https://github.com/olofk/serv/blob/master/zephyr/boards/riscv/service/service.dts#L18) to something larger, and also set the memsize parameter to the same value when you build the servant soc (e.g. `fusesoc run --target=yourtarget servant --memsize=65536`)

But I just remembered now that there is a bug in Zephyr that I forgot to report. Zephyr doesn't correctly handle RISC-V implementations without the `m` extension. So until that is fixed, you also need to edit [this line](https://github.com/zephyrproject-rtos/zephyr/blob/master/cmake/compiler/gcc/target_riscv.cmake#L13) and change `32ima` to `32i`

Let me know if it works

somhi, 2021-01-29:
I could compile without errors with both ways you told me.  I just left service.dts with 0xFFFF value.
Running the philosophers demo either in verilator or in cyc1000 just showed the demo description, but not the philosophers eating, thining, ...

I changed also 32ima to 32i in file fusesoc/zephyr/cmake/compiler/gcc/target_riscv.cmake, compiled again, and no success.
See below my commands. I changed 4096 with 8192 because I was getting the error "assert len(bindata) < 4*nwords".
When compiling in quartus I got a Warning (10850): Verilog HDL warning at servant_ram_quartus.sv(43): number of words (8192) in memory file does not match the number of elements in the address range [0:16383] 

```
cd ~/bin/fusesoc/zephyr/samples/philosophers/
west build -p auto -b service   
cd ~/bin/fusesoc
python3 $SERV/sw/makehex.py zephyr/samples/philosophers/build/zephyr/zephyr.bin 8192 > phil.hex
fusesoc run --target=verilator_tb servant --uart_baudrate=57600 --firmware=./phil.hex --memsize=65536
fusesoc run --target=cyc1000 servant --memfile=./phil.hex --memsize=65536
```

somhi, 2021-02-03:
Any ideas ?

JadanPoll, 2026-04-17:
This is the build script i use https://github.com/JadanPoll/makerchip-vizinteract/blob/main/StandAlone/zephyr_install.txt
I test my demos by running this in w3schools and importing the generated .bin
https://github.com/JadanPoll/makerchip-vizinteract/blob/main/StandAlone/Standalone_SERV_ZEPHYR.html