olofk/serv #50: Verilator Lint Fail - UNUSED

soumilheble, 2021-04-24:
The latest checkout (cfb779d3) of serv is failing verilator lint. Following is the output:
```
soumilheble@IH95-LWSTATION:~/Projects/serv$ fusesoc run --target=lint serv
INFO: Preparing ::serv:1.0.2
INFO: Setting up project

INFO: Building simulation model
verilator -f serv_1.0.2.vc -Wall
%Warning-UNUSED: ../src/serv_1.0.2/rtl/serv_params.vh:2:3: Parameter is not used: 'CSR_MSCRATCH'
                                                         : ... In instance serv_rf_top.cpu.decode
    2 |   CSR_MSCRATCH = 2'b00,
      |   ^~~~~~~~~~~~
                 ../src/serv_1.0.2/rtl/serv_decode.v:53:34: ... note: In file included from serv_decode.v
                 ... Use "/* verilator lint_off UNUSED */" and lint_on around source to disable this message.
%Warning-UNUSED: ../src/serv_1.0.2/rtl/serv_params.vh:3:3: Parameter is not used: 'CSR_MTVEC'
                                                         : ... In instance serv_rf_top.cpu.decode
    3 |   CSR_MTVEC    = 2'b01,
      |   ^~~~~~~~~
                 ../src/serv_1.0.2/rtl/serv_decode.v:53:34: ... note: In file included from serv_decode.v
%Warning-UNUSED: ../src/serv_1.0.2/rtl/serv_params.vh:4:3: Parameter is not used: 'CSR_MEPC'
                                                         : ... In instance serv_rf_top.cpu.decode
    4 |   CSR_MEPC     = 2'b10,
      |   ^~~~~~~~
                 ../src/serv_1.0.2/rtl/serv_decode.v:53:34: ... note: In file included from serv_decode.v
%Warning-UNUSED: ../src/serv_1.0.2/rtl/serv_params.vh:5:3: Parameter is not used: 'CSR_MTVAL'
                                                         : ... In instance serv_rf_top.cpu.decode
    5 |   CSR_MTVAL    = 2'b11;
      |   ^~~~~~~~~
                 ../src/serv_1.0.2/rtl/serv_decode.v:53:34: ... note: In file included from serv_decode.v
%Warning-UNUSED: ../src/serv_1.0.2/rtl/serv_params.vh:2:3: Parameter is not used: 'CSR_MSCRATCH'
                                                         : ... In instance serv_rf_top.cpu.rf_if
    2 |   CSR_MSCRATCH = 2'b00,
      |   ^~~~~~~~~~~~
                 ../src/serv_1.0.2/rtl/serv_rf_if.v:45:30: ... note: In file included from serv_rf_if.v
%Warning-UNUSED: ../src/serv_1.0.2/rtl/serv_params.vh:3:3: Parameter is not used: 'CSR_MTVEC'
                                                         : ... In instance serv_rf_top.cpu.rf_if
    3 |   CSR_MTVEC    = 2'b01,
      |   ^~~~~~~~~
                 ../src/serv_1.0.2/rtl/serv_rf_if.v:45:30: ... note: In file included from serv_rf_if.v
%Error: Exiting due to 6 warning(s)
make: *** [Makefile:16: Vserv_rf_top.mk] Error 1
ERROR: Failed to build ::serv:1.0.2 : '['make', '-j', '24', 'Vserv_rf_top.mk']' exited with an error: 2
```

* serv_decode.v is not using any of the localparams. I haven't looked at the source yet but removing the header should fix the warning.
* serv_rf_if.v is however using 2 of the 4 params.

**Potential Fix:**
```
--- a/data/verilator_waiver.vlt
+++ b/data/verilator_waiver.vlt
@@ -2,6 +2,10 @@
 // Bits [1:0] in i_ibus_rdt are not used at all
 lint_off -rule UNUSED -file "*/serv_top.v" -lines 51
 
-//Some bits in the instruction word are not used in serv_decode but it's easier
-//to just send in the whole word than picking out bits
+// Some bits in the instruction word are not used in serv_decode but it's easier
+// to just send in the whole word than picking out bits
 lint_off -rule UNUSED -file "*/serv_decode.v" -lines 6
+
+// CSR localparams (SCRATCH, MTVEC) not used in 
+// serv_rf_if.v
+lint_off -rule UNUSED -file "*/serv_rf_if.v" -lines 45

diff --git a/rtl/serv_decode.v b/rtl/serv_decode.v
index dfd4123..f5846ee 100644
--- a/rtl/serv_decode.v
+++ b/rtl/serv_decode.v
@@ -52,8 +52,6 @@ module serv_decode
    output wire              o_rd_csr_en,
    output wire              o_rd_alu_en);
 
-`include "serv_params.vh"
-
    reg [4:0] opcode;
    reg [2:0] funct3;
    reg               op20;
```

**Tools**
```
fusesoc --version
1.12.0
```
```
Verilator 4.200 2021-03-12 rev v4.200-16-gf0d66453

Copyright 2003-2021 by Wilson Snyder.  Verilator is free software; you can
redistribute it and/or modify the Verilator internals under the terms of
either the GNU Lesser General Public License Version 3 or the Perl Artistic
License Version 2.0.

See https://verilator.org for documentation

Summary of configuration:
  Compiled in defaults if not in environment:
    SYSTEMC            = 
    SYSTEMC_ARCH       = 
    SYSTEMC_INCLUDE    = /opt/systemc-2.3.3/include
    SYSTEMC_LIBDIR     = /opt/systemc-2.3.3/lib-linux64
    VERILATOR_ROOT     = /opt/verilator/share/verilator
    SystemC system-wide = 0

Environment:
    MAKE               = 
    PERL               = 
    SYSTEMC            = 
    SYSTEMC_ARCH       = 
    SYSTEMC_INCLUDE    = /opt/systemc-2.3.3/include
    SYSTEMC_LIBDIR     = /opt/systemc-2.3.3/lib-linux64
    VERILATOR_ROOT     = 
    VERILATOR_BIN      = 
```

olofk, 2021-04-26:
Thanks! I'm using a much older verilator and hadn't noticed. I decided to do another fix instead and just kill off serv_params.h since it was barely used anymore. Could you check if the latest commit solves the problem?

soumilheble, 2021-04-26:
The latest commit fixed it (a5c1c8d). Thanks!

```
soumilheble@IH95-LWSTATION:~/Projects/serv/fusesoc_libraries/serv$ cd fusesoc_libraries/serv; git log | head
bash: cd: fusesoc_libraries/serv: No such file or directory
commit a5c1c8ddc44f83fa358e9a444a7eec2dc12ba483
Author: Olof Kindgren <olof.kindgren@gmail.com>
Date:   Mon Apr 26 17:04:18 2021 +0200

    Kill off serv_params.vh
```
```
soumilheble@IH95-LWSTATION:~/Projects/serv$ fusesoc run --target=lint serv
INFO: Preparing ::serv:1.0.2
INFO: Setting up project

INFO: Building simulation model
verilator -f serv_1.0.2.vc -Wall
INFO: Running
```

closed 2021-04-26 by soumilheble