olofk/serv #62: Declare variables/nets before referenced

hakan-demirli, 2021-09-18:
Modelsim tends to get quite angry when it sees a variable that is not declared.

serv_decode.v:   "op_or_opimm"
serv_immdec.v:  "signbit"
serv_top.v:         "dbus_rdt"
Can we move them before they are referenced?

This has happened after https://github.com/olofk/serv/commit/6e802cb9bc19cefeae5d7e60d75bd6b9edb7230a

zeeshanrafique23, 2021-09-18:
Hmm, thanks for letting us know. I am amazed that Verilator doesn't catch this.

olofk, 2021-09-18:
Thanks for reporting. Yes, this should be fixed

@zeeshanrafique23 the problem is that this is totally valid to do in Verilog but ModelSim (and a few more tools) don't accept it. I have fixed this issue in SERV (and many other projects) before but it's easy to miss when code gets rewritten

zeeshanrafique23, 2021-09-18:
Ah my bad, I wasn't aware of this thing before. There is one more signal `dbus_ack` in `serv_top.v` that should also be declared before they referenced, same as `dbus_rdt`.

zeeshanrafique23, 2021-09-20:
@olofk Just made corrections, let me know if further changes are needed.

zeeshanrafique23, 2021-10-04:
@hakan-demirli this is resolved. Thanks for letting us know.

closed 2021-10-04 by hakan-demirli