olofk/serv #64: RV32E configuration

developandplay, 2021-10-16:
Hi Olof,
Thanks for creating serv.
It's the smallest RISC-V CPU I could find so far.
Small enough that it almost fits on the Speed Tang Nano. (It's just off by 100 or so LUTs.)
I saw that you posted on the RISC-V forums about RV32E for serv. Therefore I was just curious whether you implemented such a configuration.
If not I would be happy to add it if you provide me with some pointers.
I'm using the CPU with LiteX so there is probably some room to save a few LUTs there as well. (Without UART the SoC would fit but that wouldn't be too useful of a system😂).
Best wishes
Martin

developandplay, 2021-11-13:
I didn't realise how many corners SERV already cuts compared to LiteX.
Now that I read up more on it I think probably it would be an interesting idea consider getting Servant running on Gowin.
My idea for this project was basically to have a cheap, beginner friendly SoC with a RISC-V CPU to demonstrate how accessible computer architecture has become.
LiteX has really delivered on the simplicity side so I'm curious to see how fusesoc+servant compares.
Do you have good pointers on getting started for someone like me that knows LiteX pretty well?

I'll also look into cutting down UART or other components that LiteX generates.
However so far what Florent has told me is that size has not been as core of a concern for LiteX.

gregdavill, 2022-12-30:
I was thinking about this recently when packing serv into the GF180 MPW0. On FPGA targets the removal of 16*32bit registers isn't huge, typically because both configurations fit in the same BRAM blocks. 
But on an ASIC target the savings are bigger, especially if using DFF based RAM.

olofk, 2023-01-01:
I have thought about RV32E as well, but I think there's another way that can save even more memory, namely by memory-mapping the RF. We already do this in Subservient and the Serving SoClet. This alone saves us quite a bit of overhead by only needing one SRAM instead of two. But there's an added benefit in that we can also reclaim any unused registers as memory. So if we have a program that only use the first four registers, we can use the remaining 32*28 bytes as data/program memory.

I actually proposed a new extension, tentatively called Zrfinram, at the RISC-V Summit last month. You can find the video here https://award-winning.me/serv-32-bit-is-the-new-8-bit/ We don't really need a special extension for hand-coded assembly programs, but perhaps the compiler/linker could do some clever stuff if it knows which regs that are never used.