olofk/serv #72: No tool was supplied on command line or found in 'servant' core description

gojimmypi, 2021-12-23:
When running `fusesoc run --target=ulx3s servant` for the for the `ULX3S`, I see this error:

`ERROR: No tool was supplied on command line or found in 'servant' core description` 

```
gojimmypi@DESKTOP(WSL): /mnt/c/workspace ()

0 $  mkdir fusesoc-ulx3s
gojimmypi@DESKTOP(WSL): /mnt/c/workspace ()

0 $  cd fusesoc-ulx3s
gojimmypi@DESKTOP(WSL): /mnt/c/workspace/fusesoc-ulx3s ()

0 $  fusesoc library add serv https://github.com/olofk/serv
WARNING: The CAPI1 core file description format is deprecated and will be removed in the next major version of FuseSoC. Please port your core files to the CAPI2 format.
fusesoc.conf
INFO: Cloning library into fusesoc_libraries/serv
Cloning into 'fusesoc_libraries/serv'...
remote: Enumerating objects: 2277, done.
remote: Counting objects: 100% (518/518), done.
remote: Compressing objects: 100% (281/281), done.
remote: Total 2277 (delta 307), reused 366 (delta 227), pack-reused 1759
Receiving objects: 100% (2277/2277), 5.11 MiB | 4.01 MiB/s, done.
Resolving deltas: 100% (1429/1429), done.
gojimmypi@DESKTOP(WSL): /mnt/c/workspace/fusesoc-ulx3s ()

0 $  fusesoc library add fusesoc_cores https://github.com/fusesoc/fusesoc-cores
WARNING: The CAPI1 core file description format is deprecated and will be removed in the next major version of FuseSoC. Please port your core files to the CAPI2 format.
fusesoc.conf
INFO: Cloning library into fusesoc_libraries/fusesoc_cores
Cloning into 'fusesoc_libraries/fusesoc_cores'...
remote: Enumerating objects: 670, done.
remote: Counting objects: 100% (221/221), done.
remote: Compressing objects: 100% (154/154), done.
remote: Total 670 (delta 97), reused 175 (delta 55), pack-reused 449
Receiving objects: 100% (670/670), 144.79 KiB | 1.57 MiB/s, done.
Resolving deltas: 100% (241/241), done.
gojimmypi@DESKTOP(WSL): /mnt/c/workspace/fusesoc-ulx3s ()

0 $  fusesoc run --target=ulx3s servant
WARNING: The CAPI1 core file description format is deprecated and will be removed in the next major version of FuseSoC. Please port your core files to the CAPI2 format.
ERROR: No tool was supplied on command line or found in 'servant' core description
```
yet it works fine for the `tinyfpga-bx`:

```
gojimmypi@DESKTOP(WSL): /mnt/c/workspace/fusesoc-ulx3s ()

0 $  fusesoc run --target=tinyfpga_bx servant
WARNING: The CAPI1 core file description format is deprecated and will be removed in the next major version of FuseSoC. Please port your core files to the CAPI2 format.
INFO: Preparing fusesoc:utils:generators:0.1.7
INFO: Preparing ::serv:1.1.0
INFO: Preparing ::servant:1.1.0
INFO: Generating ::servant-tinyfpga_bx_pll:1.1.0

F_PLLIN:    16.000 MHz (given)
F_PLLOUT:   32.000 MHz (requested)
F_PLLOUT:   32.000 MHz (achieved)

...[snip]...

Info: [ 27230,  27986) |********************+
Info: [ 27986,  28742) |**************+
Info: [ 28742,  29498) |****************************************+
icepack servant_1.1.0_next.asc servant_1.1.0.bin
INFO: Running
```

The problem seems to be the definitions of boards missing a `PCF` file type, such as the [ulx3s](https://github.com/fusesoc/blinky/blob/c041e4bb898e9ca79a20b70146ac939799adb3f7/blinky.core#L246):

```
  ulx3s:
    files:
      - ulx3s/ulx3s_v20.lpf : {file_type : LPF, copyto : ulx3s_v20.lpf}
      - ulx3s/blinky_ulx3s.v : {file_type : verilogSource}
```

As the `zcu102 ` also fails:

```
0 $  fusesoc run --target=zcu102 servant
WARNING: The CAPI1 core file description format is deprecated and will be removed in the next major version of FuseSoC. Please port your core files to the CAPI2 format.
ERROR: No tool was supplied on command line or found in 'servant' core description
```
... and others also fail:

```
gojimmypi@DESKTOP(WSL): /mnt/c/workspace/fusesoc-ulx3s ()

0 $  fusesoc run --target=eclypse_z7 servant
WARNING: The CAPI1 core file description format is deprecated and will be removed in the next major version of FuseSoC. Please port your core files to the CAPI2 format.
ERROR: No tool was supplied on command line or found in 'servant' core description
gojimmypi@DESKTOP(WSL): /mnt/c/workspace/fusesoc-ulx3s ()

0 $  fusesoc run --target=fomu-pvt servant
WARNING: The CAPI1 core file description format is deprecated and will be removed in the next major version of FuseSoC. Please port your core files to the CAPI2 format.
ERROR: No tool was supplied on command line or found in 'servant' core description
gojimmypi@DESKTOP(WSL): /mnt/c/workspace/fusesoc-ulx3s ()
```

But note the `icebreaker`, with a [PCF](https://github.com/fusesoc/blinky/blob/c041e4bb898e9ca79a20b70146ac939799adb3f7/blinky.core#L149)

```
  icebreaker:
    files: [icebreaker/pinout.pcf : {file_type : PCF}]
```
is successful:

```
0 $  fusesoc run --target=icebreaker servant
WARNING: The CAPI1 core file description format is deprecated and will be removed in the next major version of FuseSoC. Please port your core files to the CAPI2 format.
INFO: Preparing fusesoc:utils:generators:0.1.7
INFO: Preparing ::serv:1.1.0
INFO: Preparing ::servant:1.1.0
INFO: Generating ::servant-icebreaker_pll:1.1.0

F_PLLIN:    12.000 MHz (given)
F_PLLOUT:   16.000 MHz (requested)
F_PLLOUT:   15.938 MHz (achieved)
```

I also tried the `blinky` directories that contained the [ulx3s files](https://github.com/fusesoc/blinky/tree/master/ulx3s), but all with the same error:

```
git clone https://github.com/fusesoc/blinky.git fusesoc-blinky
cd fusesoc-blinky/cd ulx3s
fusesoc run --target=ulx3s servant
```

```
fusesoc library add serv https://github.com/olofk/serv
fusesoc library add fusesoc_cores https://github.com/fusesoc/fusesoc-cores
fusesoc run --target=ulx3s servant
```

```
cd ..
fusesoc library add fusesoc_cores https://github.com/fusesoc/fusesoc-cores
fusesoc library add serv https://github.com/olofk/serv
fusesoc run --target=ulx3s servant

fusesoc run --target=tinyfpga_bx servant
fusesoc run --target=ulx3s servant
fusesoc run --target=ulx3s
```

Of course, I'm left wondering about the `WARNING: The CAPI1 core file description format is deprecated and will be removed in the next major version of FuseSoC. Please port your core files to the CAPI2 format.`.

KinzaQamar, 2021-12-24:
Greetings @gojimmypi !
Take a look at these lines https://github.com/olofk/serv/blob/28953fec4c02314f17f78c5666915a43851995fe/servant.core#:~:text=ulx3s_85%3A,toplevel%3A%20servant_ecp5 in servant.core.
1-You are targeting the fileset not the target through the command.Rewrite the command as  fusesoc run --target=ulx3s_85 servant
2-Make sure to build one of the tool mentioned under the target.By default, it uses diamond.

gojimmypi, 2021-12-24:
Hello @KinzaQamar and thank you for your suggestions!

TL;DR: it works! :)

my `servant.core` is located in `fusesoc_libraries/serv` from the above-mentioned project directory initialization.

sure enough, I was only looking at the `filesets` section. I found the `targets` you mentioned, and tada! That's Where diamond is found.

I understood the error, but did not realize that's where to edit it. 

As a suggestion to @olofk perhaps it would be a good idea to write something to the console:

`Using tool from fusesoc_libraries/serv/servant.core: diamond; see section targets:ulx3s_85`

I see that `trellis` is another option defined for the `ulx3s_85` but when I try to specify that with `--tool=trellis` I still get the same error:

```
0 $  fusesoc run --tool=trellis --target=ulx3s servant
WARNING: The CAPI1 core file description format is deprecated and will be removed in the next major version of FuseSoC. Please port your core files to the CAPI2 format.
ERROR: Failed to determine work root. Could not resolve target
```

And so here, the fact that I am _still_ targeting the  `filesets`  section, and they keyword `ulx3s` was _found_ but not in the desired `targets` section, perhaps a more intuitive error would be good here?

`WARNING: Found filesets=ulx3s but did you mean to specify a targets=[value] in your fusesoc_libraries/serv/servant.core?`

Further, note that my `servant.core` contains `CAPI=2:` at the top, contrary to the `CAPI1` warning.

I thought perhaps I could just change my default like this:

```
  ulx3s_85:
    default_tool: trellis
    description : ULX3S 85k version
    filesets : [mem_files, soc, ulx3s]
    parameters : [memfile, memsize]
    tools:
      diamond:
        part : LFE5U-85F-6BG381C
      trellis:
        nextpnr_options : [--package, CABGA381, --85k]
    toplevel: servant_ecp5
```

but I still saw the same error:

```
0 $  fusesoc run --target=ulx3s servant
WARNING: The CAPI1 core file description format is deprecated and will be removed in the next major version of FuseSoC. Please port your core files to the CAPI2 format.
ERROR: No tool was supplied on command line or found in 'servant' core description
```

(again, I didn't realize they keyword section problem of `filesets` vs `targets`)

Perhaps something as simple as:

`Error: keyword ulx3s not defined in targets in ./fusesoc_libraries/serv/servant.core`

Anyhow, it WORKS! 👍 

Changing the default as shown above, and I am able to build for the `ULX3S`:

```
fusesoc run --target=ulx3s_85 servant
```

Curiously, even when specifying the `--target=ulx3s_85` I was not able to specify the `--tool=trellis` like this:

```
0 $  fusesoc run --target=ulx3s_85 servant --tool=trellis
WARNING: The CAPI1 core file description format is deprecated and will be removed in the next major version of FuseSoC. Please port your core files to the CAPI2 format.
INFO: Preparing ::serv:1.1.0
INFO: Preparing ::servant:1.1.0
usage: fusesoc run servant_1.1.0 [-h] [--RISCV_FORMAL] [--SERV_CLEAR_RAM] [--memfile MEMFILE] [--memsize MEMSIZE] [--arch ARCH]
                                 [--output_format OUTPUT_FORMAT] [--yosys_as_subtool YOSYS_AS_SUBTOOL] [--makefile_name MAKEFILE_NAME]
                                 [--yosys_template YOSYS_TEMPLATE] [--nextpnr_options NEXTPNR_OPTIONS] [--yosys_synth_options YOSYS_SYNTH_OPTIONS]
fusesoc run servant_1.1.0: error: unrecognized arguments: --tool=trellis
```
It is apparently order-dependent, as this works:

```
fusesoc run --tool=trellis --target=ulx3s_85 servant
```

I undid my changes to the file and sure enough, I was still able to build for the ULX3S!! Woohoo!!

Thanks so much for your help! Cheers.