olofk/serv #73: Possible incorrect `ret` behaviour using default `serv_rf_top` instantiation

rschlaikjer, 2022-01-07:
I have a simple instantiation of the `serv_rf_top` module connected directly to a block RAM that doubles as firmware ROM and a small RAM. When running a test program, the CPU appears to execute correctly until it reaches a `ret` instruction and attempts to load the x1 register to read the return address. When it does so, it appears to actually load x1 * 2, resulting in a bogus PC address and undefined behaviour.

The relevant instructions of the minimal test case executing below are:
```
01000010 <func0(unsigned long, unsigned long)>:
 1000010:       00050793                mv      a5,a0
 1000014:       00008067                ret
...
void main() {
 100019c:       e75ff0ef                jal     ra,1000010 <func0(unsigned long, unsigned long)>
 10001a0:       fff00793                li      a5,-1
```

Checking the trace as we execute instruction 0x100019c, we can see on the formal interface that we are correctly writing the return address 0x10001a0 to the link register a1.
[image]

However, a few instructions later at 0x1000014, we then read a1 back but this time we get a different result, 0x02000340, which happens to be our _expected_ x1 register value but left shifted by one.
[image]

Is this a user error on my part? Is there any other information that would be useful to debug this behaviour? Unfortunately I cannot directly share the testbench I am using to replicate this error, but if necessary could create a SSCCE.

olofk, 2022-01-08:
Thanks for reporting. Easiest for me would be to see the VCD if possible. If not, I would at least like to see the dbus/ibus signals. I have had several previous bug reports that were caused by seemingly harmless bus behavior. If that doesn't show anything, an SSCCE would be useful. But before doing that it would be good to know if this problem exists in older versions since I have done some recent refactoring. Trying with SERV 1.1.0 could be worth a shot

rschlaikjer, 2022-01-10:
Here is a repo that contains a reproduction of the issue: https://github.com/rschlaikjer/serv-test

To generate a trace, `mkdir build && cmake .. && make serv_test_sim && ./serv_test_sim` should be enough. 

At t=4114 we see the value of 0x01'00'008c written to x0, at t=7358 we see 0x02'00'01'18 read back.

olofk, 2022-01-12:
I see the problem. The ack signal is not lowered correctly according to the wishbone spec. This problems exists for both ibus and dbus. The following patch would make the BFM in test_top behave correctly

```
diff --git a/test_top.v b/test_top.v
index f6a11c4..6a17979 100644
--- a/test_top.v
+++ b/test_top.v
@@ -159,16 +159,14 @@ end
 integer i;
 always @(posedge i_clk) begin
     // Serv IBus
-    serv_ibus_ack <= 0;
+    serv_ibus_ack <= serv_ibus_cyc & !serv_ibus_ack;
     if (serv_ibus_cyc) begin
         serv_ibus_rdt <= serv_rom[serv_ibus_adr[$clog2(512)-1+2:2]];
-        serv_ibus_ack <= 1;
     end
 
     // Serv DBus
-    serv_dbus_ack <= 1'b0;
+    serv_dbus_ack <= serv_dbus_cyc & !serv_dbus_ack;
     if (serv_dbus_cyc) begin
-        serv_dbus_ack <= 1'b1;
 
         // Handle reads
         case (serv_dbus_adr[31:24])
```

Another way to protect SERV  against incoming stray ack signals (without fixing the root cause) is to do like this

```
diff --git a/test_top.v b/test_top.v
index f6a11c4..00b29b7 100644
--- a/test_top.v
+++ b/test_top.v
@@ -102,7 +102,7 @@ serv_rf_top #(
    .o_ibus_adr  (serv_ibus_adr),
    .o_ibus_cyc  (serv_ibus_cyc),
    .i_ibus_rdt  (serv_ibus_rdt),
-   .i_ibus_ack  (serv_ibus_ack),
+   .i_ibus_ack  (serv_ibus_ack & serv_ibus_cyc),
 
    .o_dbus_adr  (serv_dbus_adr),
    .o_dbus_dat  (serv_dbus_dat),
@@ -110,7 +110,7 @@ serv_rf_top #(
    .o_dbus_we   (serv_dbus_we ),
    .o_dbus_cyc  (serv_dbus_cyc),
    .i_dbus_rdt  (serv_dbus_rdt),
-   .i_dbus_ack  (serv_dbus_ack),
+   .i_dbus_ack  (serv_dbus_ack & serv_dbus_cyc),
 
    // Extension
    .o_ext_rs1       (),
```

rschlaikjer, 2022-01-15:
Ah, a classic user error :facepalm: 

> I have had several previous bug reports that were caused by seemingly harmless bus behavior

I guess I'm not the only fool kicking around on here. May be worth adding a "common problems" section to the README if you end up answering this (or similar) problems too often.

Thanks for your help!

closed 2022-01-15 by rschlaikjer