olofk/serv #83: Does SERV still work with Python 2.7? Version 3.6 not detected

gojimmypi, 2022-07-23:
ok, there are a bunch of wonkiness things going on here (WSL, I know I have python 3.x installed, etc.)

But perhaps `pip install fusesoc` could detect problems and return a more friendly message?

Here's my out-of-box experience this morning using `pip install fusesoc`:

```
gojimmypi:~
$ pip install fusesoc
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Collecting fusesoc
  Downloading https://files.pythonhosted.org/packages/cd/4a/e8564fa709494a62f8de718473c76f8fadb6b5ef1b043dea1b547d0c06d0/fusesoc-1.11.0.tar.gz (101kB)
    100% |████████████████████████████████| 102kB 1.7MB/s
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-dUM5YF/fusesoc/setup.py", line 52, in <module>
        python_requires=">=3.5, <4",
      File "/usr/lib/python2.7/dist-packages/setuptools/__init__.py", line 128, in setup
        _install_setup_requires(attrs)
      File "/usr/lib/python2.7/dist-packages/setuptools/__init__.py", line 123, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 513, in fetch_build_eggs
        replace_conflicting=True,
      File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 774, in resolve
        replace_conflicting=replace_conflicting
      File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1057, in best_match
        return self.obtain(req, installer)
      File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1069, in obtain
        return installer(requirement)
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 580, in fetch_build_egg
        return cmd.easy_install(req)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 698, in easy_install
        return self.install_item(spec, dist.location, tmpdir, deps)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 724, in install_item
        dists = self.install_eggs(spec, download, tmpdir)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 909, in install_eggs
        return self.build_and_install(setup_script, setup_base)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 1177, in build_and_install
        self.run_setup(setup_script, setup_base, args)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 1163, in run_setup
        run_setup(setup_script, args)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 253, in run_setup
        raise
      File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
        self.gen.throw(type, value, traceback)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 195, in setup_context
        yield
      File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
        self.gen.throw(type, value, traceback)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 166, in save_modules
        saved_exc.resume()
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 141, in resume
        six.reraise(type, exc, self._tb)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 154, in save_modules
        yield saved
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 195, in setup_context
        yield
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 250, in run_setup
        _execfile(setup_script, ns)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 44, in _execfile
        code = compile(script, filename, 'exec')
      File "/tmp/easy_install-W_oojy/setuptools_scm-7.0.5/setup.py", line 20
        def scm_version() -> str:
                          ^
    SyntaxError: invalid syntax

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-dUM5YF/fusesoc/
You are using pip version 19.0.2, however version 20.3.4 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
gojimmypi:~
$
```

Here are some version details:

```
gojimmypi:~
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.5 LTS
Release:        18.04
Codename:       bionic
gojimmypi:~
$ python --version
Python 2.7.17
gojimmypi:~
$ python3 --version
Python 3.6.9
gojimmypi:~
$
```

KinzaQamar, 2022-07-23:
Hi, 
Try to follow the steps here: 
https://fusesoc.readthedocs.io/en/stable/user/installation.html
Hope it resolves the issue

gojimmypi, 2022-07-23:
Hi @KinzaQamar and thanks for your suggestions.

I tried the `pip3` on the `readthedocs` link instead of the README file, but it appears python had already worked that out (I've since been  trying to get this working). 

```
C:\workspace\myfusesoc>pip3 install --upgrade --user fusesoc
Requirement already up-to-date: fusesoc in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (1.12.0)
Requirement already satisfied, skipping upgrade: pyparsing in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from fusesoc) (3.0.9)
Requirement already satisfied, skipping upgrade: pyyaml in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from fusesoc) (6.0)
Requirement already satisfied, skipping upgrade: simplesat>=0.8.0 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from fusesoc) (0.8.2)
Requirement already satisfied, skipping upgrade: edalize>=0.2.3 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from fusesoc) (0.4.0)
Requirement already satisfied, skipping upgrade: ipyxact>=0.2.3 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from fusesoc) (0.3.2)
Requirement already satisfied, skipping upgrade: attrs>=17.4.0 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from simplesat>=0.8.0->fusesoc) (21.4.0)
Requirement already satisfied, skipping upgrade: okonomiyaki>=0.16.6 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from simplesat>=0.8.0->fusesoc) (1.3.2)
Requirement already satisfied, skipping upgrade: six>=1.10.0 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from simplesat>=0.8.0->fusesoc) (1.16.0)
Requirement already satisfied, skipping upgrade: Jinja2>=3 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from edalize>=0.2.3->fusesoc) (3.1.2)
Requirement already satisfied, skipping upgrade: distro; python_version >= "3.8" in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from okonomiyaki>=0.16.6->simplesat>=0.8.0->fusesoc) (1.7.0)
Requirement already satisfied, skipping upgrade: zipfile2>=0.0.12 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from okonomiyaki>=0.16.6->simplesat>=0.8.0->fusesoc) (0.0.12)
Requirement already satisfied, skipping upgrade: jsonschema>=2.5.1 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from okonomiyaki>=0.16.6->simplesat>=0.8.0->fusesoc) (4.7.2)
Requirement already satisfied, skipping upgrade: MarkupSafe>=2.0 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from Jinja2>=3->edalize>=0.2.3->fusesoc) (2.1.1)
Requirement already satisfied, skipping upgrade: importlib-resources>=1.4.0; python_version < "3.9" in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from jsonschema>=2.5.1->okonomiyaki>=0.16.6->simplesat>=0.8.0->fusesoc) (5.9.0)
Requirement already satisfied, skipping upgrade: pyrsistent!=0.17.0,!=0.17.1,!=0.17.2,>=0.14.0 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from jsonschema>=2.5.1->okonomiyaki>=0.16.6->simplesat>=0.8.0->fusesoc) (0.18.1)
Requirement already satisfied, skipping upgrade: zipp>=3.1.0; python_version < "3.10" in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from importlib-resources>=1.4.0; python_version < "3.9"->jsonschema>=2.5.1->okonomiyaki>=0.16.6->simplesat>=0.8.0->fusesoc) (3.8.1)
WARNING: You are using pip version 19.2.3, however version 22.1.2 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.
```

It still might be a good idea to detect python versions more gracefully with `pip`  vs `pip3` ? Perhaps just update the README and/or refer to the `readthedocs` link. 

The `upgrade` gave essentially the same output as above:  `pip3 install --upgrade fusesoc`

The Windows instructions I assume are very similar, but still listed at a TODO on the `readthedocs` link:

[image]

gojimmypi, 2022-07-23:
oops, I confused this WSL issue with my Windows issue.

Here's the result in WSL, using `pip3` resolves, but there's a new error:

```
$ pip3 install --upgrade --user fusesoc
Collecting fusesoc
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/69/65/e5902fea9bac81e56e4f5d81e4a2ba497244a160f3ffa7df59e7e1d3f030/fusesoc-1.12.0-py3-none-any.whl (63kB)
    100% |████████████████████████████████| 71kB 1.4MB/s
Collecting edalize>=0.2.3 (from fusesoc)
  Downloading https://files.pythonhosted.org/packages/ef/3a/ebaf6e01a797f0ac4896ec4b688b8b39ad22d7919b0d1470c60e591408ad/edalize-0.4.0.tar.gz (337kB)
    100% |████████████████████████████████| 337kB 1.8MB/s
Collecting ipyxact>=0.2.3 (from fusesoc)
  Downloading https://files.pythonhosted.org/packages/e6/4b/3e7485acab5b762f8b417db2fc890c12853a52bef4b278a33ef6919b4808/ipyxact-0.3.2.tar.gz (201kB)
    100% |████████████████████████████████| 204kB 2.5MB/s
Collecting simplesat>=0.8.0 (from fusesoc)
  Downloading https://files.pythonhosted.org/packages/1f/79/aca2b68a5fb5f2a3183bf2ac0f09fb05dc6b4ce859dc1aa52670c44866a8/simplesat-0.8.2-py2.py3-none-any.whl (246kB)
    100% |████████████████████████████████| 256kB 1.5MB/s
Collecting pyyaml (from fusesoc)
  Downloading https://files.pythonhosted.org/packages/b3/85/79b9e5b4e8d3c0ac657f4e8617713cca8408f6cdc65d2ee6554217cedff1/PyYAML-6.0-cp36-cp36m-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_12_x86_64.manylinux2010_x86_64.whl (603kB)
    100% |████████████████████████████████| 604kB 1.6MB/s
Collecting pyparsing (from fusesoc)
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/6c/10/a7d0fa5baea8fe7b50f448ab742f26f52b80bfca85ac2be9d35cdd9a3246/pyparsing-3.0.9-py3-none-any.whl (98kB)
    100% |████████████████████████████████| 102kB 4.1MB/s
Collecting Jinja2>=3 (from edalize>=0.2.3->fusesoc)
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/20/9a/e5d9ec41927401e41aea8af6d16e78b5e612bca4699d417f646a9610a076/Jinja2-3.0.3-py3-none-any.whl (133kB)
    100% |████████████████████████████████| 143kB 3.9MB/s
Collecting six>=1.10.0 (from simplesat>=0.8.0->fusesoc)
  Cache entry deserialization failed, entry ignored
  Using cached https://files.pythonhosted.org/packages/d9/5a/e7c31adbe875f2abbb91bd84cf2dc52d792b5a01506781dbcf25c91daf11/six-1.16.0-py2.py3-none-any.whl
Collecting okonomiyaki>=0.16.6 (from simplesat>=0.8.0->fusesoc)
  Downloading https://files.pythonhosted.org/packages/81/df/1ec79aeb85dd3e6551bba3d7a2bfa8591e1bafc69e7d405fdf6132f65bf1/okonomiyaki-1.3.2-py3-none-any.whl (2.8MB)
    100% |████████████████████████████████| 2.8MB 434kB/s
Collecting attrs>=17.4.0 (from simplesat>=0.8.0->fusesoc)
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/be/be/7abce643bfdf8ca01c48afa2ddf8308c2308b0c3b239a44e57d020afa0ef/attrs-21.4.0-py2.py3-none-any.whl (60kB)
    100% |████████████████████████████████| 61kB 2.5MB/s
Collecting MarkupSafe>=2.0 (from Jinja2>=3->edalize>=0.2.3->fusesoc)
  Cache entry deserialization failed, entry ignored
  Using cached https://files.pythonhosted.org/packages/fc/d6/57f9a97e56447a1e340f8574836d3b636e2c14de304943836bd645fa9c7e/MarkupSafe-2.0.1-cp36-cp36m-manylinux1_x86_64.whl
Collecting zipfile2>=0.0.12 (from okonomiyaki>=0.16.6->simplesat>=0.8.0->fusesoc)
  Downloading https://files.pythonhosted.org/packages/60/ad/d6bc08f235b66c11bbb76df41b973ce93544a907cc0e23c726ea374eee79/zipfile2-0.0.12-py2.py3-none-any.whl (44kB)
    100% |████████████████████████████████| 51kB 1.5MB/s
Collecting jsonschema>=2.5.1 (from okonomiyaki>=0.16.6->simplesat>=0.8.0->fusesoc)
  Downloading https://files.pythonhosted.org/packages/e0/d9/05587ac378b9fd2c352c6f024f13240168365bd753a7e8007522b7025267/jsonschema-4.0.0-py3-none-any.whl (69kB)
    100% |████████████████████████████████| 71kB 2.6MB/s
Collecting importlib-metadata; python_version < "3.8" (from jsonschema>=2.5.1->okonomiyaki>=0.16.6->simplesat>=0.8.0->fusesoc)
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/a0/a1/b153a0a4caf7a7e3f15c2cd56c7702e2cf3d89b1b359d1f1c5e59d68f4ce/importlib_metadata-4.8.3-py3-none-any.whl
Collecting pyrsistent!=0.17.0,!=0.17.1,!=0.17.2,>=0.14.0 (from jsonschema>=2.5.1->okonomiyaki>=0.16.6->simplesat>=0.8.0->fusesoc)
  Downloading https://files.pythonhosted.org/packages/6c/19/1af501f6f388a40ede6d0185ba481bdb18ffc99deab0dd0d092b173bc0f4/pyrsistent-0.18.0-cp36-cp36m-manylinux1_x86_64.whl (117kB)
    100% |████████████████████████████████| 122kB 2.0MB/s
Collecting typing-extensions>=3.6.4; python_version < "3.8" (from importlib-metadata; python_version < "3.8"->jsonschema>=2.5.1->okonomiyaki>=0.16.6->simplesat>=0.8.0->fusesoc)
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/45/6b/44f7f8f1e110027cf88956b59f2fad776cca7e1704396d043f89effd3a0e/typing_extensions-4.1.1-py3-none-any.whl
Collecting zipp>=0.5 (from importlib-metadata; python_version < "3.8"->jsonschema>=2.5.1->okonomiyaki>=0.16.6->simplesat>=0.8.0->fusesoc)
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/bd/df/d4a4974a3e3957fd1c1fa3082366d7fff6e428ddb55f074bf64876f8e8ad/zipp-3.6.0-py3-none-any.whl
Building wheels for collected packages: edalize, ipyxact
[... cut ...]
25.3.6. Executing AST frontend in derive mode using pre-parsed AST for module `\serv_rf_top'.
Parameter \RESET_PC = 0
Parameter \WITH_CSR = 1
Generating RTLIL representation for module `$paramod\serv_rf_top\RESET_PC=0\WITH_CSR=1'.
../src/servant_1.0.2-r1/servant/servant_timer.v:0: ERROR: Can't find object for defparam `RESET_STRATEGY`!
Makefile:8: recipe for target 'servant_1.0.2-r1.json' failed
make: *** [servant_1.0.2-r1.json] Error 1
ERROR: Failed to build ::servant:1.0.2-r1 : '['make']' exited with an error: 2
gojimmypi:~/workspace/mywslfusesoc
$
```

gojimmypi, 2022-07-23:
update: SUCCESS~

I compared `fusesoc library list` on WSL to my Windows/DOS version.

Ran this: `fusesoc library add servant https://github.com/olofk/serv`

```
$ fusesoc library list
Name          : Location                        : Sync type : Sync URI                                 : Auto sync
fusesoc-cores : fusesoc_libraries/fusesoc-cores : git       : https://github.com/fusesoc/fusesoc-cores : y
gojimmypi:~/workspace/mywslfusesoc
$ fusesoc library add servant https://github.com/olofk/serv
INFO: Cloning library into fusesoc_libraries/servant
Cloning into 'fusesoc_libraries/servant'...
remote: Enumerating objects: 2471, done.
remote: Counting objects: 100% (583/583), done.
remote: Compressing objects: 100% (187/187), done.
remote: Total 2471 (delta 458), reused 417 (delta 393), pack-reused 1888
Receiving objects: 100% (2471/2471), 5.27 MiB | 8.05 MiB/s, done.
Resolving deltas: 100% (1585/1585), done.
```

and... TADA!

```
gojimmypi:~/workspace/mywslfusesoc
$ fusesoc run --target=tinyfpga_bx servant
INFO: Preparing fusesoc:utils:generators:0.1.7
INFO: Preparing ::serv:1.1.0
INFO: Preparing ::servant:1.1.0
INFO: Generating ::servant-tinyfpga_bx_pll:1.1.0

F_PLLIN:    16.000 MHz (given)
F_PLLOUT:   32.000 MHz (requested)
F_PLLOUT:   32.000 MHz (achieved)

FEEDBACK: SIMPLE
F_PFD:   16.000 MHz
F_VCO: 1024.000 MHz

[ .. snip .. ]

Info: Max delay posedge wb_clk_$glb_clk -> <async>: 4.05 ns

Info: Slack histogram:
Info:  legend: * represents 3 endpoint(s)
Info:          + represents [1,3) endpoint(s)
Info: [ 15256,  15948) |+
Info: [ 15948,  16640) |+
Info: [ 16640,  17332) |
Info: [ 17332,  18024) |************+
Info: [ 18024,  18716) |*************+
Info: [ 18716,  19408) |**+
Info: [ 19408,  20100) |**************+
Info: [ 20100,  20792) |*************+
Info: [ 20792,  21484) |*****+
Info: [ 21484,  22176) |******+
Info: [ 22176,  22868) |************+
Info: [ 22868,  23560) |******************************+
Info: [ 23560,  24252) |*******************************************************+
Info: [ 24252,  24944) |*********************************+
Info: [ 24944,  25636) |**************************+
Info: [ 25636,  26328) |*****************************************+
Info: [ 26328,  27020) |*****+
Info: [ 27020,  27712) |***********+
Info: [ 27712,  28404) |********************+
Info: [ 28404,  29096) |************************************************************
icepack servant_1.1.0.asc servant_1.1.0.bin
INFO: Running
gojimmypi:~/workspace/mywslfuseso
```

success!

KinzaQamar, 2022-07-23:
Awesome !

olofk, 2022-07-24:
Great to see you got it working. Thanks for the detailed reports of your struggles. Thought I had seen the last Python2 issue, but it seems python2 just refuses to die. You're probably right that it's better to explicitly mention `pip3` over `pip` for systems that still use python2 as the default version.

The other error (`ERROR: Can't find object for defparam `RESET_STRATEGY`!`) has been fixed by adding a newer version of SERV (with the error fixed) to the `fusesoc-cores` library