olofk/serv #86: fusesoc problems with python on Windows - python3 vs python

gojimmypi, 2022-07-23:
similar to https://github.com/olofk/serv/issues/83 in WSL, the Windows version of `fusesoc` seems to not properly detect which version of python to use.  

Here in Windows, `python` is Python version 3.6 and `python3` is not defined, causing a `fusesoc` error:

```
C:\workspace\myfusesoc>fusesoc run --target=tinyfpga_bx servant
←[1;37mINFO: Preparing fusesoc:utils:generators:0.1.7←[0m
←[1;37mINFO: Downloading fusesoc/fusesoc-generators from github←[0m
←[1;37mINFO: Preparing ::serv:1.1.0←[0m
←[1;37mINFO: Preparing ::servant:1.1.0←[0m
←[1;37mINFO: Generating ::servant-tinyfpga_bx_pll:1.1.0←[0m
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.
←[1;31mERROR: Setup failed : "python3 C:\Users\gojimmypi\.cache\fusesoc\fusesoc_utils_generators_0.1.7\icepll.py C:\Users\gojimmypi\.cache\fusesoc\generated\servant-tinyfpga_bx_pll_1.1.0\tinyfpga_bx_pll_input.yml" exited with an error code. See stderr for details.←[0m

C:\workspace\myfusesoc>python3 --version
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.

C:\workspace\myfusesoc>python --version
Python 3.8.1
```

KinzaQamar, 2022-07-23:
Hi, 
Check which version of fusesoc you are on through this command ```fusesoc --version``` .
Version 1.10 of fusesoc supports python2.7. I am not sure but my best guess is the installed version of fusesoc is not compatible with the installed version of python.

Try this command ``sudo pip3 install --upgrade fusesoc``. As you already have fusesoc, this command will upgrade the older version to a newer one.
If the above command still gives an error, try to install the python3 dependencies then run the above command.

Hope it will do something.

gojimmypi, 2022-07-23:
Hi @KinzaQamar - I appreciate your assistance.

I was able to eventually get `fusesoc` installed, per `pip` vs `pip3 noted in https://github.com/olofk/serv/issues/83

The `fusesoc`` version on Windows reports:

```
C:\workspace\myfusesoc>fusesoc --version
1.12.0
```

but I still see this error:

```
C:\workspace\myfusesoc>fusesoc run --target=tinyfpga_bx servant
←[1;37mINFO: Preparing fusesoc:utils:generators:0.1.7←[0m
←[1;37mINFO: Preparing ::serv:1.1.0←[0m
←[1;37mINFO: Preparing ::servant:1.1.0←[0m
←[1;37mINFO: Generating ::servant-tinyfpga_bx_pll:1.1.0←[0m
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.
←[1;31mERROR: Setup failed : "python3 C:\Users\gojimmypi\.cache\fusesoc\fusesoc_utils_generators_0.1.7\icepll.py C:\Users\gojimmypi\.cache\fusesoc\generated\servant-tinyfpga_bx_pll_1.1.0\tinyfpga_bx_pll_input.yml" exited with an error code. See stderr for details.←[0m

```

KinzaQamar, 2022-07-23:
Hi,
Do you have python3 dependencies installed already?

gojimmypi, 2022-07-24:
Hi @KinzaQamar - which dependencies are you referring to? The `pip3 install fusesoc`? Yes.

Here's the output from various commands. I created a new `C:\workspace\myfusesoc2` Windows directory to start fresh:

```
C:\workspace\myfusesoc2>fusesoc core show servant
CORE INFO
Name:      ::servant:1.1.0
Core root: fusesoc_libraries\serv

Targets:
ac701           : <No description>
alhambra        : Open-hardware iCE40HX4K FPGA board
arty_a7_35t     : <No description>
ax309           : XILINX Spartan-6 XC6SLX9 FPGA Development Board
chameleon96     : Chameleon96 (Arrow 96 CV SoC Board)
cmod_a7_35t     : <No description>
cyc1000         : cyc1000 FPGA board
de0_nano        : <No description>
de10_nano       : Terasic DE10 Nano Kit
deca            : DECA development kit by Arrow / Terasic
default         : <No description>
ebaz4205        : EBAZ4205 'Development' Board
go_board        : <No description>
icebreaker      : <No description>
icestick        : <No description>
icesugar        : iCE40UP5K Development Board by MuseLab
lint            : <No description>
lx9_microboard  : LX9 Microboard
nexys_2_1200    : <No description>
nexys_2_500     : <No description>
nexys_a7        : <No description>
orangecrab_r0.2 : OrangeCrab R0.2
pipistrello     : Saanlima pipistrello
sim             : <No description>
sockit          : SoCKit development kit by Arrow / Terasic
tinyfpga_bx     : <No description>
ulx3s_85        : ULX3S 85k version
upduino2        : <No description>
verilator_tb    : <No description>
zcu106          : Zynq UltraScale+ MPSoC ZCU106 Evaluation Kit

C:\workspace\myfusesoc2>fusesoc library list
Name          : Location                        : Sync type : Sync URI                                 : Auto sync
fusesoc_cores : fusesoc_libraries\fusesoc_cores : git       : https://github.com/fusesoc/fusesoc-cores : y
serv          : fusesoc_libraries\serv          : git       : https://github.com/olofk/serv            : y

C:\workspace\myfusesoc2>fusesoc run --target=tinyfpga_bx servant
←[1;37mINFO: Preparing fusesoc:utils:generators:0.1.7←[0m
←[1;37mINFO: Preparing ::serv:1.1.0←[0m
←[1;37mINFO: Preparing ::servant:1.1.0←[0m
←[1;37mINFO: Generating ::servant-tinyfpga_bx_pll:1.1.0←[0m
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.
←[1;31mERROR: Setup failed : "python3 C:\Users\gojimmypi\.cache\fusesoc\fusesoc_utils_generators_0.1.7\icepll.py C:\Users\gojimmypi\.cache\fusesoc\generated\servant-tinyfpga_bx_pll_1.1.0\tinyfpga_bx_pll_input.yml" exited with an error code. See stderr for details.←[0m

C:\workspace\myfusesoc2> fusesoc --version
1.12.0

C:\workspace\myfusesoc2>pip3 install --upgrade fusesoc
Requirement already up-to-date: fusesoc in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (1.12.0)
Requirement already satisfied, skipping upgrade: simplesat>=0.8.0 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from fusesoc) (0.8.2)
Requirement already satisfied, skipping upgrade: edalize>=0.2.3 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from fusesoc) (0.4.0)
Requirement already satisfied, skipping upgrade: ipyxact>=0.2.3 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from fusesoc) (0.3.2)
Requirement already satisfied, skipping upgrade: pyparsing in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from fusesoc) (3.0.9)
Requirement already satisfied, skipping upgrade: pyyaml in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from fusesoc) (6.0)
Requirement already satisfied, skipping upgrade: six>=1.10.0 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from simplesat>=0.8.0->fusesoc) (1.16.0)
Requirement already satisfied, skipping upgrade: okonomiyaki>=0.16.6 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from simplesat>=0.8.0->fusesoc) (1.3.2)
Requirement already satisfied, skipping upgrade: attrs>=17.4.0 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from simplesat>=0.8.0->fusesoc) (21.4.0)
Requirement already satisfied, skipping upgrade: Jinja2>=3 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from edalize>=0.2.3->fusesoc) (3.1.2)
Requirement already satisfied, skipping upgrade: distro; python_version >= "3.8" in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from okonomiyaki>=0.16.6->simplesat>=0.8.0->fusesoc) (1.7.0)
Requirement already satisfied, skipping upgrade: zipfile2>=0.0.12 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from okonomiyaki>=0.16.6->simplesat>=0.8.0->fusesoc) (0.0.12)
Requirement already satisfied, skipping upgrade: jsonschema>=2.5.1 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from okonomiyaki>=0.16.6->simplesat>=0.8.0->fusesoc) (4.7.2)
Requirement already satisfied, skipping upgrade: MarkupSafe>=2.0 in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from Jinja2>=3->edalize>=0.2.3->fusesoc) (2.1.1)
[... cut ...]
Requirement already satisfied, skipping upgrade: zipp>=3.1.0; python_version < "3.10" in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from importlib-resources>=1.4.0; python_version < "3.9"->jsonschema>=2.5.1->okonomiyaki>=0.16.6->simplesat>=0.8.0->fusesoc) (3.8.1)
WARNING: You are using pip version 19.2.3, however version 22.1.2 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.
```

Although it might be nice to get this working on Windows, this issue is not nearly as important since I have `fusesoc` working on WSL as noted in https://github.com/olofk/serv/issues/83#issuecomment-1193150975

olofk, 2022-07-24:
Hmm... this one is a bit tricky. I would had assumed that there would be a python3 symlink/alias on every platform. Especially since we have seen other places where things go haywire because python is linked to python2.

For this particular problem, I see a manual workaround for now. Edit `C:\workspace\myfusesoc2\fusesoc_libraries\fusesoc-cores\fusesoc_utils\generators-0.1.7.core` and change the interpreter on line 50 from `python3` to `python`. Some context: The servant SoC is using a generator (generators are kind of plugins in the FuseSoC world) to create a PLL with desired frequency settings. This generator is implemented in python, and the core description file for the generator (that you are about to edit) explicitly tells FuseSoC to launch this generator with python3.