olofk/serv #87: Dockerfile for SERV environment

sbates130272, 2022-08-01:
Hi

I would like to propose we add a docker folder with a Dockerfile in it that contains a Ubuntu based container with all the packages and other assorted goodies needed to build the SERV CPU and run tests. I would be happy to work on a PR for this based on the contents of the CI workflows already checked in.

Is this something that others would consider useful? I would plan to default to checking out the HEAD of the master branch but allow users to pass tags or SHAs into the build command. I think, in time, we could also extent it to include some of the open-source EDA tools needed to build bitstreams for some of the target FPGA boards etc.

Feedback welcome!

olofk, 2022-08-02:
That does sound like a good idea. My biggest question would be how people expect to use such an image. As I'm not using Docker regularly myself I'm not sure what the expected work flow looks like

sbates130272, 2022-08-03:
@olofk what I typically do is create the Dockerfile and update the README at the top-level to include a reference to it and tell the user that all the necessary software is already installed. Once they spin up the image they can just run it and have all the tools they need ready to be used.

I'll try and put something together for review in the next few days.