olofk/serv #93: Unable to run Compliance Tests

AshvinVaidyanathan, 2022-11-14:
```
ls: cannot access '/home/ashvin/thesis/workspace/riscv-arch-test/riscv-target/serv/device/rv32i_m': No such file or directory

============================ VARIABLE INFO ==================================
ROOTDIR: /home/ashvin/thesis/workspace/riscv-arch-test [origin: file]
WORK: /home/ashvin/thesis/workspace/riscv-arch-test/work [origin: file]
TARGETDIR: /home/ashvin/thesis/workspace/riscv-arch-test/riscv-target [origin: command line]
RISCV_TARGET: serv [origin: command line]
XLEN: 32 [origin: file]
RISCV_DEVICE: I [origin: command line]
=============================================================================

make -j1 \
	RISCV_TARGET=serv \
	RISCV_DEVICE=I \
	run -C /home/ashvin/thesis/workspace/riscv-arch-test/riscv-test-suite/rv32i_m/I
make[1]: Entering directory '/home/ashvin/thesis/workspace/riscv-arch-test/riscv-test-suite/rv32i_m/I'
../../Makefile.include:25: *** Cannot find '/home/ashvin/thesis/workspace/riscv-arch-test/riscv-target/serv/device/rv32i_m/I/Makefile.include`. Check that RISCV_TARGET and RISCV_DEVICE are set correctly..  Stop.
make[1]: Leaving directory '/home/ashvin/thesis/workspace/riscv-arch-test/riscv-test-suite/rv32i_m/I'
make: *** [Makefile:84: simulate] Error 2
```
As far as I understand I am supposed to point the test to the files built by fusesoc, but there is no such directory once the build is finished. Can someone point me out to where I am going wrong?

KinzaQamar, 2022-11-14:
Hi,

Have you set your WORKSPACE and SERV paths?

AshvinVaidyanathan, 2022-11-15:
Thanks a lot. That worked. Although the first test (ADD) failed. Is this expected?

KinzaQamar, 2022-11-15:
Try to remove work directory inside riscv-arch-test directory before every run.

closed 2022-11-15 by AshvinVaidyanathan