olofk/serv #97: %Warning: /sw/zephyr_hello.hex:0: $readmem file not found

DerakhshanAli, 2023-04-23:
hello.
first of all, thanks for sharing the SERV project.
facing problem : 
Command : fusesoc run --target=verilator_tb servant --uart_baudrate=57600 --firmware=$SERV/sw/zephyr_hello.hex
Result : Loading RAM from /sw/zephyr_phil.hex
%Warning: /sw/zephyr_phil.hex:0: $readmem file not found

How can i solve this Warning?

i think, it's because this warning ,I **dont** get [***** Booting Zephyr OS zephyr-v1.14.1-4-gc7c2d62513fe *****
Hello World! service]

thanks.

closed 2023-04-23 by DerakhshanAli

YiminGao0113, 2023-12-16:
Seems like this thread is already closed but I didn't find the answer to this question.. Could someone tell me how I can resolve this Warning, my program got stuck here..
[image]
Thanks.

olofk, 2023-12-16:
It looks like you haven't set the $SERV environment variable. The --firmware option specify a path to a hex file.