olofk/serv #102: Just a question building Zephyr demos.

hvegh, 2023-08-01:
Hi Olof,

Thanks for serv!
Yesterday I managed to get the serv core running with the hello world example, running 72Mhz using on a Tang Nano 9K and getting the Hello World @ 259200 baud. 

Next step would be change the Hello World application to something useful. 

Never used **west** before, any quick commands that will generate a servant risc-v binary from source?

H

hvegh, 2023-08-01:
So far following Zephyr 101 - Creating Your Own Project:
```
west init -m https://github.com/olofk/serv.git
west update
pip install -r  zephyr/scripts/requirements.txt

cd ~/.local
wget https://github.com/zephyrproject-rtos/sdk-ng/releases/download/v0.16.1/zephyr-sdk-0.16.1_linux-x86_64_minimal.tar.xz
tar xvf zephyr-sdk-0.16.1_linux-x86_64_minimal.tar.xz
cd zephyr-sdk-0.16.1
./setup.sh 
# select riscv64 toolchain?
# install hosttools

cd ~/prj/zephyrprj/zephyr
west build -p always -b litex_vexriscv samples/hello_world
CMake Error: The source "/home/henk/prj/zephyr-project/zephyr/samples/hello_world/CMakeLists.txt" does not match the source
```
Zephyr install: award winning rabbit hole.

hvegh, 2023-08-01:
retry:

```
cd ~/prj/fusesoc/
west init
west config manifest.path fusesoc_libraries/serv
west update
cd zephyr/samples/hello_world/
west build -b service
```

burb:
```
-- Configuring done (3.5s)
CMake Warning (dev) at /home/henk/prj/fusesoc/zephyr/cmake/linker/ld/target.cmake:33 (add_custom_command):
  Policy CMP0116 is not set: Ninja generators transform DEPFILEs from
  add_custom_command().  Run "cmake --help-policy CMP0116" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  /home/henk/prj/fusesoc/zephyr/CMakeLists.txt:682 (configure_linker_script)
This warning is for project developers.  Use -Wno-dev to suppress it.
-- Generating done (0.0s)
-- Build files have been written to: /home/henk/prj/fusesoc/zephyr/samples/hello_world/build
-- west build: building application
[1/104] Preparing syscall dependency handling
...
[2/76] Building ASM object ...
fusesoc/fusesoc_libraries/serv/zephyr/soc/riscv/servant/soc_irq.S: Assembler messages:
fusesoc/fusesoc_libraries/serv/zephyr/soc/riscv/servant/soc_irq.S:28: Error: unrecognized opcode `csrrc t1,mip,t0'
fusesoc/fusesoc_libraries/serv/zephyr/soc/riscv/servant/soc_irq.S:47: Error: unrecognized opcode `csrr t0,mcause'
```

hvegh, 2023-08-04:
Disected the gcc options on the commandline:
Question now is how and where to enable these instructions on the Zephyr toolchain...

```
.local/zephyr-sdk-0.16.1/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc 
-DBUILD_VERSION=zephyr-v2.4.0 
-DKERNEL 
-D_FORTIFY_SOURCE=2 
-Os -imacros  -ffreestanding -fno-common 
-g -mabi=ilp32 -march=rv32ima 
-Wall -Wformat 
-Wformat-security 
-Wno-format-zero-length 
-Wno-main -Wno-pointer-sign 
-Wpointer-arith 
-Wno-address-of-packed-member 
-Wno-unused-but-set-variable
-Werror=implicit-int 
-fno-asynchronous-unwind-tables 
-fno-pie 
-fno-pic 
-fno-strict-overflow 
-fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/henk/prj/fusesoc/zephyr/samples
....
```

olofk, 2023-08-07:
This is an unfortunate change in the toolchains. You will need an older version of zephyr-sdk. It looks like I'm using version 0.11.4. There are probably newer ones that works as well, though. Not to self that this should be added to the instructions as well.

olofk, 2023-08-07:
Ah! I just remembered one more thing. There's a bug in Zephyr that causes some parts of Zephyr to be built with the M extension (mul/div instructions) even if you tell Zephyr that your CPU doesn't support it. This has been fixed in later versions of Zephyr, but for now you need to make the following change manually in Zephyr.

```diff
diff --git a/cmake/compiler/gcc/target_riscv.cmake b/cmake/compiler/gcc/target_riscv.cmake
index 88c7a901f3..ed517538f2 100644
--- a/cmake/compiler/gcc/target_riscv.cmake
+++ b/cmake/compiler/gcc/target_riscv.cmake
@@ -9,7 +9,7 @@ if(CONFIG_64BIT)
     list(APPEND TOOLCHAIN_C_FLAGS -mcmodel=medany)
 else()
     string(CONCAT riscv_mabi  "i" ${riscv_mabi} "32")
-    string(CONCAT riscv_march ${riscv_march} "32ima")
+    string(CONCAT riscv_march ${riscv_march} "32i")
 endif()
 
 if(CONFIG_FPU)

```

hvegh, 2023-10-27:
Hi Olof,

Sorry for the late response, I finaly managed to get it working.
Since then I moved over to the Litex framework, this has exactly what I need.

closed 2023-10-27 by hvegh