olofk/serv #118: Python package clash

shareefj, 2024-02-09:
I'm trying to run the riscof tests in `verif/` but there seems to be a pip dependency issue between what `fusesoc` requires and the version required for `riscof`.

If I have setup a venv and installed `fusesoc`, running `pip install riscof` results in the following error:

```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
fusesoc 2.3 requires pyyaml>=6.0, but you have pyyaml 5.2 which is incompatible.
jsonschema2md 1.1.0 requires PyYAML<7.0.0,>=6.0.0, but you have pyyaml 5.2 which is incompatible.
```

However, if I add all dependencies to a single `requierments.txt`, then it seems to install with no issues.  I am very far from a pip expert so thought I'd just report this and see if anyone had any pointers.  Is pip able to resolve dependency clashes if everything is installed at once?

Cheers.

olofk, 2024-02-13:
Yes. I actually have this problem myself. It's the riscv-config package (which is a dependency of riscof) which has a strict dependency on pyyaml 5.2. I'm actually using a modified local version of that package to get around the problem (clone the repo https://github.com/riscv-software-src/riscv-config, change `setup.py` to say `pyyaml>=5.2` instead of `pyyaml==5.2`, and then do a "development install" by running `pip install --user -e .` from the riscv-config repo).

I also had to do a similar hack for riscv_isac to allow for newer versions of pyelftools in the requirements.txt there.

Generally, I do wonder if the whole riscof framework is alive at all tbh. Not that I'm the most active maintainer myself, but I haven't seen any activity for ages in the riscof or riscv-config repos

shareefj, 2024-02-13:
Yeah, having just gone through the pain of setting it up on a core, the documentation is well out of date and there are so many little bugs and issues it feels like it should be scrapped.  I've just emailed Neel who is a repo owner of the arch tests to check. 

But anyway, I think you can avoid this issue by adding all your Python dependencies to a single requirements.txt file and update your documentation.  At least that worked for me in my limited testing.