olofk/serv #133: mbedTLS on SERV

m42uko, 2024-08-20:
Hi,

I am currently trying to evaluate the performance of doing some crypto on SERV, but I'm running into some issues with Zephyr on SERV.

I wrote the attached example (main.c) that just initializes the RSA library according to their example, and then prints a string in a loop (no actual en-/decryption happens yet).

Yet, when I run the example (both on real HW and in simulation through fusesoc+verilator), the entire system hangs indefinitely (real HW) or resets (sim). If you remove the mbedtls init calls, the loop runs fine.

I have been trying to troubleshoot this a bit myself already, but I was hoping you might have some more experience analyzing these issues.

I attached the following files: [rsa-test-zephyr-serv.zip](https://github.com/user-attachments/files/16678149/rsa-test-zephyr-serv.zip)
* main.c: My test program (don't forget to enable the mbedtls RSA + entropy configs)
* log.txt: Output from verilator
* rsa-test-zephyr-serv.hex: The hex file I tested with
* run.sh: My run command for the simulation
* rsa-test-zephyr-serv.tar.xz: the entire application directory for zephyr (including configs and build results that produced the described output)

Let me know if you have any ideas on how to approach this / if you find out anything. Thank you for your help!

PS: I have not yet tried running the example program on any other RISC-V target. That's still on my TODO list.

olofk, 2024-08-23:
I haven't found time to look at this issue, but I did just fix an issue that made the regression tests fail. Probably unrelated, but could be worth trying with the latest git version.

Also, during my debugging sessions, I have develeoped some rudimentary tools that can make life slightly easier.

On the trace branch, I have added a WIP tracing module that shows register values and what opcode that was just executed. This makes things slightly easier to follow when doing waveform debugging.

You can also switch on the --trace_pc switch for simulations. This will generate a `trace.bin` that contains the PC trace. Together with [this questionable python hack](https://github.com/user-attachments/files/16727983/trays.tar.gz) you can do more analysis of what the CPU is doing. By default it will generate a trace with annotations of all jumps if you run it with `python trays.py /path/to/elf/file /path/to/trace.bin`

Another thing I typically do is to take the assembler listing of the program you want to run and check for illegal instructions or attempted accesses to unimplemented CSR registers. To save size, SERV has zero protection from these things and will happily try to execute instructions that it doesn't have the slightest clue about how to actually run. Looking quickly at the file you sent me, there is a read from mhartid. That might be totally fine, but worth taking an extra look at. I also see a wfi instruction + a couple of zext.b (which a quick googling tells me is the same as andi??) that could be wort investigating.

m42uko, 2024-09-10:
I didn't have much time to work on this, but I at least looked at your suggestions.

**1. `mhartid` and `wfi`**

There is only one access to `mhartid`, and it's conveniently in the same block as the `wfi`: The bootup section for unused cores (which we should never hit):
[image]

I'm pretty sure everything here is fine as we only have one CPU core, and even if the read from `mhartid` returned a core other than core 0, I'd expect a more severe error than a crash late in the program.

**2. `zext.b`**

I looked at the disassembly. `zext.b` produces the same assembly code as an `andi` with immediate `255`.

**3. Your latest updates**

I pulled the latest `main` branch today, and it sadly does not fix the issue.

**.**

I hope I'll have some more time to look into this in the upcoming days, but no promises :/

closed 2024-11-08 by m42uko

m42uko, 2024-12-19:
For anyone wondering, and for me to confirm that this is not a problem with SERV, here's what the problem was: I ran out of stack memory in the main thread and overwrote critical sections of OS memory. Doh!

The solution is to either increase the stack memory for the main/init process in the .config, or to create a second thread with more stack memory. Here's my working example utilizing the second method: [decrypt.zip](https://github.com/user-attachments/files/18193706/decrypt.zip) Takes about 25 minutes to decrypt 128 bytes (@25 MHz core clock), but it works :D

olofk, 2024-12-19:
Very cool. Thanks for sharing.

And that exact problem of running out of stack space has happened to me as well. It's an absolute pain to debug and looks really similar to a hardware bug. Would be nice to have some better way to detect this problem, at least in simulation.